RaspAP/raspap-webgui #1: Internet connection via wlan1 & 3G to wlan0

MikeOne86, 2015-08-04:
Are there any settings that the Internet connection via WLAN and 3G? Or is it planned?
Best Regards Mike

webnull, 2015-12-13:
You could take a look at https://github.com/webnull/raspap-webgui
This is a refactored version of RaspAP, here when you setup 3G interface using eg. Network Manager you could use it as a internet source for your hotspot interface.

billz, 2017-06-06:
Hi @MikeOne86, circling back to this issue as I have more time to devote to the project. There are related requests #77 and #78 to offer greater control for multiple wlan interfaces. I'm prototyping a UI that will hopefully capture much of what's needed on this front. Thanks.

billz, 2019-04-13:
Updated the [FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#interfaces) to address this.

closed 2019-04-13 by billz