RaspAP/raspap-webgui #13: Use ssl

jrmhaig, 2016-06-26:
Lighttpd should be configured to use https rather than http.

billz, 2017-06-03:
@jrmhaig I see this as a key feature going forward. One approach would be to generate a self-signed certificate with OpenSSL for the **raspberrypi.local** domain. The Avahi mDNS daemon (`avahi-daemon`) is a default package on  fresh images of Raspbian Jessie, Jessie Lite and Wheezy. So with a properly issued certificate https://raspberrypi.local/ (or raspap.local, whatever) should resolve without extra software.

There are of course issues with modern browsers rejecting self-signed certificates as insecure, given that they don't chain back to a trust anchor. Some browsers (Android's default one, I believe) will throw a warning and not give the option of trusting the server's certificate.

The best way around this, that I know of, is to become a certificate authority (CA), generate the self-signed cert with proper keys and DNS names in the Subject Alternate Name (SAN) and install the CA certificate on the client. 

I'll do some testing around this with lighttpd and document it in this issue.

managai, 2019-06-29:
Hi, some updates on SSL self-signed certificate for RaspAP ?

billz, 2019-07-29:
See this [wiki page](https://github.com/billz/raspap-webgui/wiki/HTTPS--support) for steps on configuring RaspAP with HTTPS support.

billz, 2019-11-06:
Per the wiki, the big advantage with `mkcert` is it doesn't generate self-signed certificates, but rather certificates signed by your own private CA. 

**Question:** would this be a desirable option for the Quick Installer? Add a reaction if you'd like to see this added.

My thinking is the installer could use the hostname and suggest a .local certificate generated with mkcert. lighttpd would then be configured for SSL as per the wiki steps. Finally, the `rootCA.pem` file would be copied to RaspAP's web folder so it could be distributed to clients.

billz, 2019-11-07:
The quick installer can now be used to generate SSL certs with mkcert. This includes configuration of lighttpd. Append `-c` or `--cert` to the installer, like so:

`wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap --cert` 

Feedback is welcome.

billz, 2019-11-08:
Updated the readme and wiki accordingly, closing

closed 2019-11-08 by billz