RaspAP/raspap-webgui #18: RaspAP INSTALL ERROR: Unable to download files from github

janislejins, 2016-08-03:
having trouble using the installer script. 
I think there needs to be a .git at line 72... of common.sh install 
relevant section:
```
# Fetches latest files from github to webroot
function download_latest_files() {
    if [ ! -d "$webroot_dir" ]; then
        install_error "Web root directory doesn't exist"
    fi

    install_log "Cloning latest files from github"
    sudo git clone https://github.com/billz/raspap-webgui "$webroot_dir" || install_error "Unable to download files from github"
}
```

should be (?):
```
# Fetches latest files from github to webroot
function download_latest_files() {
    if [ ! -d "$webroot_dir" ]; then
        install_error "Web root directory doesn't exist"
    fi

    install_log "Cloning latest files from github"
    sudo git clone https://github.com/billz/raspap-webgui.git "$webroot_dir" || install_error "Unable to download files from github"
}
```

janislejins, 2016-08-03:
made a fork with the change and still threw same error. Never mind I'll troubleshoot on my end.

jrmhaig, 2016-08-12:
I haven't used the install scripts myself but I would guess that it is failing because `/var/www` (`$webroot_dir`) is not empty.

billz, 2016-08-17:
Thanks for the heads up, I'll check it out.

billz, 2016-08-31:
Fixed in #29.

closed 2016-08-31 by billz