RaspAP/raspap-webgui #20: 'Wireless Information' on Dashboard?

jrmhaig, 2016-08-08:
I am unsure of the purpose of the 'Wireless Information' section on the main dashboard page. For me it is (mostly) blank and I presume that it only has information when the wireless adapter is configured as a client attached to another SSID. This, however, seems to defeat the object of having an AP.
Is the intention to be able to connect to an existing wireless network and then, with a second wireless interface, provide an extension?
I am in the process of re-doing the dashboard so that, for instance, it includes information about eth0 and I am inclined to remove this section unless it has some value.

zlolz, 2016-08-08:
billz is obviously better placed to respond to the intention, but I can speak to some of your other questions:

Wireless AP <-> Wireless Client is a common use case, and was what originally brought me to raspap-webgui.  It's very useful in travelling/roaming situations for example.

You are absolutely correct that the current code doesn't serve 'wireless client' and 'wired client' equally, or cleanly.  The current structure doesn't make it easy to correct this, either.

'Second wireless interface' is certainly an option, but not a requirement, for Wireless AP <-> Wireless Client.  Many devices allow this on the *same* physical interface.  Importantly, including the RPI3.

Adding information (and configuration) of wired devices is a good start, but please maintain parity for those who need wireless client connections too.

Thanks for all your work on this -- much appreciated :)

Aside:  I ran in to similar issues for a while and maintained a large patchset before realizing my requirements (seamless roaming, always-on vpn with hotspot support etc) could not be satisfied without a ground-up rewrite.  I believe it even started with a rework of the dashboard - good luck :)

jrmhaig, 2016-08-09:
Thanks for the information and, in particular, confirmation that the Raspberry Pi 3 wireless can be configured to have multiple interfaces. I thought that may be the case but didn't know.

I have added a new issue, #21, to add support for this. This is not something I am likely to work on in the short term but someone else may pick it up.

zlolz, 2016-08-09:
No problem - I've added a bit more detail to #21.  Apologies for not being clearer that Wireless Client <-> Wireless AP is possible with the current raspap-webui.

Thanks again!

closed 2020-03-19 by stale[bot]