RaspAP/raspap-webgui #21: Support multiple wireless interfaces

jrmhaig, 2016-08-09:
Multiple wireless interfaces will allow, for example, RaspAP to be configured as a wireless extender.

It is possible with some devices to configure multiple interfaces. In particular, this can be done with the built in wireless on the Raspberry Pi 3.

zlolz, 2016-08-09:
iirc raspap-webgui will [enumerate and allow selection of](https://github.com/billz/raspap-webgui/pull/7) all available wireless interfaces so long as they have been setup outside the app.  This allows for raspap to be configured as a a repeater or extender.

(again, iirc) The dashboard shows statistics etc for wlan0 only.  Not ideal but workable as wlan0 is usually the wifi interface used for client connections in that configuration (and if not can be renamed as such in /etc/interfaces).

Pi3:  Additional (AP) interface on a Pi3 without a second *physical* interface can be accomplished by something like: `iw dev wlan0 interface add ap0 type __ap` but note that the AP channel and Client channel will need to match.

nquick, 2017-04-17:
Just wanted to say that it has been excellent and easy to setup on my Pi3 using the onboard wifi.  What I'd like to do is configure my client onto a separate wifi dongle (uses the RTL8188CUS chipset which can be a pain with hostapd configs). I notice I can't do this via the web gui. 

It would be massively helpful and make the device act like a WISP router.

billz, 2017-06-06:
See also related issues #77 #78 and #1. Clearly a hot item 😉

billz, 2019-03-04:
PR's are welcome.

closed 2019-03-04 by billz