RaspAP/raspap-webgui #25: Display signal level appropriately for how it is reported

jrmhaig, 2016-08-17:
As noted in #24, the 'Signal level' can be reported by `iwconfig` either as dBm or as a ratio (possibly always over 100, so effectively a percentage) depending on the hardware and software drivers of the wireless interface. A suitable regex at:

* https://github.com/jrmhaig/raspap-webgui/blob/20eb3b0107f5c30c2c16a2ab535b7d92c9284792/includes/dashboard.php#L42

could be used to determine which format this is and then the display at:

* https://github.com/jrmhaig/raspap-webgui/blob/20eb3b0107f5c30c2c16a2ab535b7d92c9284792/includes/dashboard.php#L102

can be changed to a single text field or bar graph as appropriate.

zlolz, 2016-08-17:
Possible alternative is using iw.

`iw $IFACE link` - shows signal, reports in dBm (only), and is nl80211-based.  

> [The old tool iwconfig, which uses Wireless Extensions interface, is deprecated and it's strongly recommended to switch to iw and nl80211.](https://wireless.wiki.kernel.org/en/users/documentation/iw)

Travelling currently and don't want to mess up my pi, else this would be a pull request :)

billz, 2019-11-05:
The project has since [transitioned](https://github.com/billz/raspap-webgui/commit/25f362dc1937ae3c4f80e3bfa082d77816cfafe2#diff-a2159bbaf42b7f61ddba68fff6623118) to using `iw`. It uses this output to determine signal level as an aggregate value, which is now displayed as 'Link Quality' with a donut graph on the dashboard.

closed 2019-11-05 by billz