RaspAP/raspap-webgui #27: RaspAP INSTALL ERROR: Cannot enable fastcgi-php for lighttpd

hoshymann, 2016-08-30:
Thanks for the great work. RaspAP seems to be exactly what I am looking for!

But I have problems getting it working.
When using the automatic installer I am getting this error message:
RaspAP INSTALL ERROR: Cannot enable fastcgi-php for lighttpd

If I follow the steps for manual installation, everything installs fine (also enabling PHP for lighttpd works), but after the final reboot, calling the IP-adress of the Pi from a PC just gives me "404 - Not Found" page.
It does not seem to make a difference, if the PC is connected with the same LAN as the Pi or directly with the wifi the Pi is hosting.

What am I missing?

jrmhaig, 2016-08-30:
I'm afraid I haven't looked much at the automatic installer so I would recommend going through the README instructions instead, as you did.

After the reboot are you able to ssh (or ping) in via the IP address?

hoshymann, 2016-08-30:
No worries, your README is great, so no problems there.

Pinging the PI sure works, I just tried it to be certain. But be aware thet the 404 message I get is no "Page not found" message from Firefox after a timeout or something. It seems to come from the Pi itself.

jrmhaig, 2016-08-30:
Which directory did you check out the code into. I cannot remember offhand whether it should be directly into `/var/www` or `/var/www/html`.

hoshymann, 2016-08-31:
Oh, ok. I checked out the code into `/var/www` as stated in the README. But I will try `/var/www/html` as soon as I get to it and keep you posted on the results.

jrmhaig, 2016-08-31:
See step 5 of https://www.raspberrypi.org/forums/viewtopic.php?t=151358&p=1030101

It appears that the default directory for lighttpd changed from `/var/www` to `/var/www/html` in the latest version of Raspbian (Jessie)

jrmhaig, 2016-08-31:
To fix this issue the installation script needs to:
* Detect the version of Raspbian
* Install RaspAP into `/var/www` or `/var/www/html` as appropriate

hoshymann, 2016-08-31:
Ok, perferct. Now it works!
Thank you for the help and the great code!

closed 2016-08-31 by hoshymann

Bennadade, 2019-04-18:
please help me. I am trying to install RaspAP and this is the error occured: 
RaspAP INSTALL ERROR: unable to restart  lighttpd