RaspAP/raspap-webgui #30: Other install folder

risoul, 2016-09-01:
I installed it into /var/www/html/raspap.
Wireless client works fine, but when trying to configure hotspot, it redirect me to a wrong page.
I searched for a base url parameter but did'nt find.

jrmhaig, 2016-09-01:
Are you viewing it at http://ip-address/raspap? And are the links dropping the raspap?

I cannot check it right now but can you try adding this line to (round about) line 67 of index.php:

```html
<base url="http://ip-address/raspap">
```

replacing `ip-address` to the address of your Pi. This isn't a complete fix but
1. it may indicate what the fix could be
2. it may be a quick fix for you

risoul, 2016-09-04:
Yes I view it at ttp://ip-address/raspap
Links on the left are working, but start and save buttons are dropping.

I temporary solved it by replacing
` <form role="form" action="/?page=hostapd_conf" method="POST">`
by
` <form role="form" action="/raspap/?page=hostapd_conf" method="POST">`

I will try your solution and let you know.

Don't you think it could be better to install it into a folder and not delete /var/www/html at install?

jrmhaig, 2016-09-11:
It looks like removing the `/` from a couple of places fixes it and this change will be merged shortly. Can you check and confirm that this works for you?

PR #35 (2016-09-11): Set form actions correctly
Fixes #30
closed 2016-09-11 by jrmhaig