RaspAP/raspap-webgui #32: Client wireless Discover

enrutador, 2016-09-06:
Congratulations for the work. I have a question:

In the old version, on the part of wifi client settings before you allow for an discover the wifis networks and select the desired. In the new version the option "Scan networks" does not appear and only works if you add the networks are defined in the wpa_supplicant.conf file.

The new version does not allow me not from the web, create the network in manual mode, but does not find a Discover Network

You're working on this feature? or you'll have to create the wireless connection manually?

Thank you

Before:

[image]

After:

[image]

jrmhaig, 2016-09-07:
It should scan each time you load the page and display everything it finds. If this is not the case then there is a bug. The networks are detected using:

```bash
sudo wpa_cli scan
sleep 3
sudo wpa_cli scan_results
```

Do your networks appear in the output from these commands? And can you check that these commands are in the sudoers file for the user www-data. This hasn't changed so I wouldn't expect this to be the problem.

One thing that springs to mind is that it will only discover networks that broadcast their ssid and so you cannot configure a hidden network. This is an oversight and I will look into fixing that.

The screenshot suggests that there is something else wrong even when you do have the network configured. The icons on the left indicate that the network is configured (in the `wpa_supplicant.conf` file) and connected. However, it is not able to identify the channel. Can you find the following lines in `includes/configure_client.php`

* https://github.com/billz/raspap-webgui/blob/master/includes/configure_client.php#L121
* https://github.com/billz/raspap-webgui/blob/master/includes/configure_client.php#L127

and change `ConvertToChannel($arrNetwork[1])` to `$arrNetwork[1]` and then see what appears under the 'Channel' column in the table.

enrutador, 2016-09-07:
First of all I must tell you that the hardware I'm using is a BEAGLEBONE Green Wireless. This machine has a 1 wireless interface but with a 2x2 Mimo Wifi / Ble. This allows me to create in the same two communications interface

1) SoftAP to connect client computers (hostapd + chillicoova -> ALL OK)
2) The wlan0 that connects the router and performs the function of default gateway

The points you mention me are:

1) Sudoers / wpa
www-data ALL = (ALL: ALL) NOPASSWD: ALL

arm root @: ~ # wpa_cli scan
Selected 'p2p-dev-wlan0' interface
okay
arm root @: ~ # sleep 3
arm root @: ~ # wpa_cli scan_results
Selected 'p2p-dev-wlan0' interface
bssid / frequency / signal level / flags / ssid

2) The network is not hidden

3) The configuration of the network interface Wifi performed in the system is as follows:

/ Etc / network / interfaces:

allow-hotplug wlan0
iface wlan0 inet dhcp
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
iface default inet dhcp

/etc/wpa_supplicant/wpa_supplicant.conf:

ctrl_interface = DIR = / var / run / wpa_supplicant GROUP = netdev

network = {
  ssid = "LOVER"
  scan_ssid = 1
  key_mgmt = WPA-PSK
  psk = "XXXXX"
  proto = WPA RSN
  pairwise = CCMP TKIP
  group = CCMP TKIP
}
4) Chances for frequency problems:

I have made changes to code:
     # $ Networks [$ arrNetwork [4]] [ 'channel'] = ConvertToChannel ($ arrNetwork [1]);
     $ Networks [$ arrNetwork [4]] [ 'channel'] = $ arrNetwork [1];

      # 'Channel' => ConvertToChannel ($ arrNetwork [1]),
     'Channel' => $ arrNetwork [1],

I reload apache2 server , but don´t show the frequency 
[image]

5) Observations
I see the new sytem Pi revision page,very interesting ¡¡  the command for discover the BBXX versio is:

https://github.com/gkaindl/beaglebone-ubuntu-scripts/blob/master/bb-show-serial.sh

Need more information , can do any test more?

jrmhaig, 2016-09-07:
I think that the different hardware is most likely the source of the problems, and possibly the software too. Are you using Raspbian? Or do you have something else that is based off Debian? I don't know how much customisation goes into Raspbian but I would think that provided you are on something Debian based, and preferably Jessie, it should be OK.

The output from `wpa_cli scan_results` suggests that it did not find any networks. This may be the case if the interface is already set up as an access point. It may also be that the scan has not completed (which is what the `sleep 3` is for) but I would be surprised if it takes more than half a second.

At the top you say that wlan0 is used for connecting to the router but the output of the wpa_cli commands suggest that the interface is p2p-dev-wlan0. I'm afraid I don't know how the 2x2 Mimo Wifi works so I don't know what the interface names should be. You could try adding `-i <interface name>` to the wpa_cli commands with some different names and see if any of them respond. You could also try it again with the access point disabled.

You could also try using `iw dev <interface> scan` to detect networks instead of `wpa_cli`. `wpa_cli` is what was being used when I started working on this project but I believe that `iw` is the new tool that should use instead. See

* https://wireless.wiki.kernel.org/en/users/documentation/iw

If this does a better job at detecting the networks it might be worth moving to that now. I'm fairly sure it works on a Raspberry Pi but I didn't get round to going through the code and changing it.

Regarding the hardware revision in the system information, this currently comes from a simple look-up table based on `/etc/cpuinfo`. The method for Beaglebone appears to be more involved and I'm not likely to look at it any time soon (not least as I don't have a Beaglebone) but if you want you could raise a separate issue containing that link.

enrutador, 2016-09-07:
Hi Joseph
Yes correct , I am using a debian Jessie:

Linux 4.4.16 arm-ti-rt-r38 # 1 SMP PREEMPT RT Sat 6 Aug 2016 1:18:53 CEST armv7l GNU / Linux

As well you say the interface is connected. With the "iw" command is fantastic, even connected gives you the scan of associated networks. I am attaching the output of the scan as the properties of the nic with "iw" command

I would like to help you change those functions and perform the test with "iw" I can go over all the code and go changing commands to see how it behaves interface wifi

I created a fork of your project on my site I can upload everything observed with my bbgw

Debug Commands:
Scan:

root@arm:~# iw dev wlan0 scan | more
BSS 00:19:77:1d:e8:94(on wlan0) -- associated
        TSF: 5293302450023 usec (61d, 06:21:42)
        freq: 2472
        beacon interval: 100 TUs
        capability: ESS Privacy ShortPreamble ShortSlotTime (0x0431)
        signal: -59.00 dBm
        last seen: 4920 ms ago
        Information elements from Probe Response frame:
        SSID: LOVER
        Supported rates: 1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0
        DS Parameter set: channel 13
        Country: ES     Environment: Indoor/Outdoor
                Channels [1 - 13] @ 20 dBm
        ERP: <no flags>
        RSN:     * Version: 1
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: PSK
                 * Capabilities: 1-PTKSA-RC 1-GTKSA-RC (0x0000)
                 * 0 PMKIDs
        Extended supported rates: 24.0 36.0 48.0 54.0*
        HT capabilities:
                Capabilities: 0x18c
                        HT20
                        SM Power Save disabled
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 8 usec (0x06)
                HT RX MCS rate indexes supported: 0-15
                HT TX MCS rate indexes are undefined
        HT operation:
                 * primary channel: 13
                 * secondary channel offset: no secondary
                 * STA channel width: 20 MHz
                 * RIFS: 0
                 * HT protection: nonmember
                 * non-GF present: 1
                 * OBSS non-GF present: 1
                 * dual beacon: 0
                 * dual CTS protection: 0
                 * STBC beacon: 0
                 * L-SIG TXOP Prot: 0
                 * PCO active: 0
                 * PCO phase: 0
        Overlapping BSS scan params:
                 * passive dwell: 20 TUs
                 * active dwell: 10 TUs
                 * channel width trigger scan interval: 300 s
                 * scan passive total per channel: 200 TUs
                 * scan active total per channel: 20 TUs
                 * BSS width channel transition delay factor: 5
                 * OBSS Scan Activity Threshold: 0.25 %
[... 281 lines cut ...]
                 * #{ managed } <= 2, #{ AP, P2P-client, P2P-GO } <= 1, #{ P2P-device } <= 1,
                   total <= 3, #channels <= 2
                 * #{ AP } <= 2, #{ P2P-device } <= 1,
                   total <= 2, #channels <= 1, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz }

                 * #{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1,
                   total <= 3, #channels <= 2
                 * #{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1,
                   total <= 3, #channels <= 1, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz }

                 * #{ managed } <= 1, #{ AP } <= 1, #{ P2P-GO } <= 1, #{ P2P-device } <= 1,
                   total <= 3, #channels <= 1
        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        Device supports TX status socket option.
        Device supports HT-IBSS.
        Device supports SAE with AUTHENTICATE command
        Device supports scan flush.
        Device supports per-vif TX power setting
        Driver supports a userspace MPM

jrmhaig, 2016-09-07:
Thanks. Something to bear in mind is that all commands that need root access will have to be added to the sudoers file exactly as it will be called which can cause problems if, for example, it requires the interface name as an argument. So if you have 

```bash
sudo iw dev wlan0 scan
```

Then this will only work for an interface called wlan0. This is a bit of a pain and I'm not sure how to get round it but there are probably ways of calling commands to dump all information for all devices. What we certainly don't want to do is give the www-data user full passwordless sudo access.

enrutador, 2016-09-08:
Hi Joseph

With awk and iwlist command I could adapt a little wpa_cli scan command output to be as much like your code structure.

`sudo iwlist wlan0 scan | awk '/ESSID/ {gsub("ESSID:\""," ",$1); /"/ gsub("\""," ",$1); printf "%s ", $1} /Address/ {printf "%s ", $5} /Frequency/ {gsub("\)","",$4); printf "%s ", $4} /802.11i/ {gsub("802.11i/","",$3); print " ", $3}'`

The output is :

`00:19:77:1D:E8:94 13  LOVER    WPA2
E2:41:36:64:50:50 1  MOVISTAR_5050    WPA2
F4:06:8D:27:FF:22 11  devolo-f4068d27ff22    WPA2
`

The only problem is if the network shows the SSDI with spaces. I upgrade my fork code for testing

Thanks

jrmhaig, 2016-09-11:
The PHP can be changed to deal with the output of iwlist instead of wpa_cli so it may not be necessary to have the complicated awk command. That can be done in PHP instead. In fact this is probably preferable if, as you say, an SSID can have spaces.

I haven't had a chance to try this out on a Raspberry Pi yet but I'll try to get round to it.

jrmhaig, 2016-09-18:
Your changes appear to work with Raspberry Pi (although I have only have a very brief look) but it needs `/sbin/iwlist wlan0 scan` added to the sudoers file in both the README and the installer script. Merge from upstream first (see https://help.github.com/articles/syncing-a-fork/) as I have made changes to make the sudoers file clearer.
I still think that for simplicity the complicated awk command should be avoided and the information picked out in the PHP.

enrutador, 2016-09-19:
Hi Joseph
Thanks for the information,now i working with de openvpn section, its interesting for me do the vpn client on the device.
Thanks

billz, 2016-11-08:
Closing this. Feel free to reopen it as needed.

closed 2016-11-08 by billz