RaspAP/raspap-webgui #39: Raspberry pi 3 support without dongle?

khromov, 2016-09-26:
RPi 3 has wifi built-in. Would this be possible to add support for this? Thanks for a great project.

jrmhaig, 2016-09-27:
Can you specify what isn't working? I hate to say "works for me" but, erm, it "works for me". :-)

pyroclasticflux, 2016-10-17:
Having similar issue (RPi3B). Dashboard tells me "interface is down" (it's not). Starting wlan0 via dashboard button does nothing. Hotspot config tells me "HostAPD is not running" (it is). Again button yields no result (get a message "Attempting to start hotspot", but besides that nothing).

Dnsmasq is behaving nicely and all the buttons work.

Works fine on an RPi1B for me with same method.

One problem seems to be that ifup is not able to read /etc/network/interfaces properly. Its a clean raspbian install and after that I only followed steps to install raspap, don't know what went wrong.

khromov, 2016-10-17:
+1 @pyroclasticflux

jrmhaig, 2016-10-18:
For a short period there was a problem with the setup script although it was quickly fixed. If, by chance, you set up the Pi while it was broken you may have a problem with your sudoers file. Can you check if you have these lines (edit with `sudo visudo`:

```
www-data ALL=(ALL) NOPASSWD:/sbin/ifdown wlan0
www-data ALL=(ALL) NOPASSWD:/sbin/ifup wlan0
www-data ALL=(ALL) NOPASSWD:/bin/cat /etc/wpa_supplicant/wpa_supplicant.conf
www-data ALL=(ALL) NOPASSWD:/bin/cp /tmp/wifidata /etc/wpa_supplicant/wpa_supplicant.conf
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli scan_results
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli scan
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli reconfigure
www-data ALL=(ALL) NOPASSWD:/bin/cp /tmp/hostapddata /etc/hostapd/hostapd.conf
www-data ALL=(ALL) NOPASSWD:/etc/init.d/hostapd start
www-data ALL=(ALL) NOPASSWD:/etc/init.d/hostapd stop
www-data ALL=(ALL) NOPASSWD:/etc/init.d/dnsmasq start
www-data ALL=(ALL) NOPASSWD:/etc/init.d/dnsmasq stop
www-data ALL=(ALL) NOPASSWD:/bin/cp /tmp/dhcpddata /etc/dnsmasq.conf
www-data ALL=(ALL) NOPASSWD:/sbin/shutdown -h now
www-data ALL=(ALL) NOPASSWD:/sbin/reboot
```

What happens if, from a command line, you try these two commands:

```
sudo -u web-data sudo /etc/init.d/hostapd start
sudo -u web-data sudo /sbin/ifup wlan0
```

Finally, can you see if there are any errors that might be relevant in `/var/log/lighttpd/error.log`. Thanks.

jrmhaig, 2016-10-21:
I don't know if this is the problem (I would be surprised if it is RPi 3 specific) but I have just tried running the installer on a newly installed and fully updated Pi and am encountering problems. I'm working on the branch https://github.com/billz/raspap-webgui/tree/improve_installer if you want to see how far I am getting.

PureGrain, 2016-12-28:
One thing I've found that may help is adding this to the bottom of the hostapd.conf for Raspi 3

`ieee80211n=1 # 802.11n support
wmm_enabled=1 # QoS support
ht_capab=[HT40][SHORT-GI-20][DSSS_CCK-40]`

billz, 2017-05-19:
Closing this as fixes from #42 and #57 have been merged. Please reopen if the issue appears on other hardware revisions, thanks.

closed 2017-05-19 by billz