RaspAP/raspap-webgui #40: lighttpd[640]: Failed to connect to non-global ctrl_ifname: (null)  error: No such file or directory

rnamdos, 2016-10-03:
Getting this error on syslog when I want to go to the "Configure client" tab. I'm not able to input my wifi information on it either.
I'm a bit confused. Are we able to connect via hostap via wifi, scan the wifi networks and then connect to a wifi network of choice?

billz, 2016-11-08:
That is the intention, correct. The latest release creates a default hostapd config with the easy installer. Perhaps you can try this and let me know if it resolves your issue?

billz, 2017-06-03:
Closing this as there's been no activity. @rnamdos please reopen if your issue persists, thanks.

closed 2017-06-03 by billz

sourabhtk37, 2017-12-20:
Having the same issue. Running raspap inside a docker container on jessie via manual installation.