RaspAP/raspap-webgui #41: Changes to sudoers file

jrmhaig, 2016-10-21:
Upgrading to the latest Raspbian it appears that the default sudoers file has changed. The file `/etc/sudoers.d/README` says:

```
#
# As of Debian version 1.7.2p1-1, the default /etc/sudoers file created on
# installation of the package now includes the directive:
# 
# 	#includedir /etc/sudoers.d
# 
# This will cause sudo to read and parse any files in the /etc/sudoers.d 
# directory that do not end in '~' or contain a '.' character.
# 
# Note that there must be at least one file in the sudoers.d directory (this
# one will do), and all files in this directory should be mode 0440.
# 
# Note also, that because sudoers contents can vary widely, no attempt is 
# made to add this directive to existing sudoers files on upgrade.  Feel free
# to add the above directive to the end of your /etc/sudoers file to enable 
# this functionality for existing installations if you wish!
#
# Finally, please note that using the visudo command is the recommended way
# to update sudoers content, since it protects against many failure modes.
# See the man page for visudo for more information.
#
```

This should make it easier to manage the permissions for www-data as a single file can be created and then simply dropped into `/etc/sudoers.d` with the correct permissions set. I am not making the change yet as some backwards compatibility should be thought about first.

3663, 2016-11-07:
Is this what's giving me a headache trying to get logged into the web GUI? Been faffing about for several hours and have gotten nowhere

jrmhaig, 2016-11-08:
This is unlikely.

What problem are you encountering? Are you asked for a username and password or are you not even seeing that? Is this a completely new installation or has this problem just started recently, possibly after a software update?

3663, 2016-11-08:
Fresh install, used the easy installer. Seems to be working mostly. 
It has created an open wireless network called raspi-webgui  which is not allowing Internet access. When connected to the network I can browse to the gui login with the specified IP address, but I cannot login. 
I have. Created new root users using visudo and have confirmed they have root access.
However I cannot find any login which will be successful to get onto the network config gui, I have also modified the raspap.conf to include these usernames.

jrmhaig, 2016-11-08:
The password for the 'admin' user should be 'secret'.
If you can log in to the Pi can you check that the wired device (eth0) is up and has an IP address? And can you ping something on the internet?

jrmhaig, 2016-11-08:
Also, can you try the following:

```
cat /proc/sys/net/ipv4/ip_forward
```

3663, 2016-11-08:
SSH into the pi allows me to ping sites and run update/upgrade so it doesn't appear that this is an issue.  The baseline pi system is getting Web access over eth0. Isn't the network supposed to be secured with the pasephrase ChangeMe as standard? 
Any attempt to login to the web GUI results in failure. Tried admin - secret with varying capital letters and punctuation.  Even tried a brute with no success. 
I have the pi setup with an assigned static IP for a previous project, not sure if this is tying it in knots. I will try with a fresh raspbian install on another card and see if it's something in my current config causing it.

jrmhaig, 2016-11-08:
There should be a copy of `raspap.php` in `/etc/raspap`, which contains the default password (see https://github.com/billz/raspap-webgui/blob/master/raspap.php).

If there is a file called `/etc/raspap/raspap.auth` then this may contain a new password if it has changed. If you find this file exists then the simplest thing would be to delete it.

The default password for the wireless should be 'ChangeMe'. From what you are saying it sounds as though the setup script isn't configuring forwarding from the wlan0 to eth0. Could you please tell me the output from:

```
cat /proc/sys/net/ipv4/ip_forward
```

I'll try going through a clean install on an up-to-date Raspbian when I get the chance but it may not be for a day or two.

3663, 2016-11-08:
Ran the script on a fresh Raspbian install, same results. The output of cat /proc/sys/net/ipv4/ip_forward is 0 . 
ls showed no file named raspap.auth and pulling the SD and checking manually in explorer shows no such file. 
Still able to ping google with no issues.

jrmhaig, 2016-11-13:
Hi @3663. Copying this file:

  https://github.com/billz/raspap-webgui/blob/fix_default_setup/config/rc.local

over `/etc/rc.local` should set up the routing on next boot. I am going to check it a little more and then create a pull request for this change, after which the setup script will copy the file in place automatically.

billz, 2017-06-02:
@jrmhaig getting back to your original point re: Raspbian update to sudoers file handling. This would indeed simplify managing permissions with www-data. For backwards compatibility, we can leverage the Raspbian version detection in the Quick Installer.

commit c6492466 (2020-03-19): Update sudoers file handling. Resolves #41. Addresses #127
closed 2020-03-29 by billz