RaspAP/raspap-webgui #44: Dashboard access admin/secret not working after fresh install

delkant, 2016-11-18:
Very Nice project!!

How can I trace errors? 

The only thing I changed from the default installation config was the path of the app to /usr/local/bin/

If I connect to the "raspi-webgui" and go to the default IP address  admin/secret won't work and I will keep getting the user/pass screen.

Please help!

jrmhaig, 2016-11-18:
Logs are written to the lighttpd log files in `/var/log/lighttpd` and you will need `sudo` to view them.

When you say you have changed the path of the app do you mean the path it asks for at the start of the automatic install? I am wondering if that is broken and it will not work if you don't put it in the default location, in which case it should probably not give an option. Can you try adding a symlink from `/etc/raspap` (the default) to `/usr/local/bin` to see if it then works.

By the way, this path is for configuration files so `/usr/local/bin` probably isn't the right place. Most of the code actually goes into `/var/www/html`.

delkant, 2016-11-18:
Thank you for the quick reply. I noticed I have no content on /etc/raspap (maybe because I changed the path on the automatic install and the folder wasn't created? )

Is there a place in the repository where I can get the files I need to put in /etc/raspap?

delkant, 2016-11-18:
this worked for me!: 

 adding a symlink from /etc/raspap (the default) to /usr/local/bin

And yes you are right about the /usr/local/bin I didn't know what this will hold before the installation, I just wanted to use this location because I am putting here all the extra apps I am using on my Raspberry.

Thank you again!!

closed 2016-11-18 by delkant