RaspAP/raspap-webgui #46: Get IP address from DHCP when configured as client

jrmhaig, 2016-11-21:
When connecting to a wireless network as a client the IP address that is allocated via DHCP should be used but currently it keeps the default 10.3.141.1. The workaround, as indicated in #45, is to remove or comment out the following lines from `/etc/dhcpcd.conf`:
```
static ip_address=10.3.141.1/24
static routers=10.3.141.1
static domain_name_server=8.8.8.8 8.8.4.4
```
This issue is related to the suggestion mentioned in issue #31.

hdavid, 2017-03-23:
hi,

Shouldn't the approach be : 
- first try to get an ip from the network via (remote) dhcp
- then only if failed (wlan ip would be 169.* then)
 - start dnsmasq
 - assign static ip

Otherwise you can endup with two DHCP on the same network, and actually break the existing network.

i am working on a project where i implemented similar feature to raspap, before deciding to use raspap itself and found this is issue. and i have implemented a (very hackish) way to slove that.
https://github.com/hdavid/ports/blob/master/install.sh
```
#!/bin/bash
inet=`hostname -I`
if [[ $inet == 169*  ]];
then
    # no ip, we are in AP mode.
    # set static ip 
    ifconfig wlan0 down
    ifconfig wlan0 192.168.1.1 netmask 255.255.255.0 up
    sudo /etc/init.d/dnsmasq start
fi
```

i am happy to contribute a pull request where this would be added to the /etc/rc.local or smth like this. let me know if it makes sense.

jrmhaig, 2017-03-23:
That might be useful as a quick fix and as this question has come up a few times then it would be useful to do this in the short term.

However, I think it would be better to have the device configured to a particular mode as suggested in #31.

SirLagz, 2017-10-30:
PR #131 enables this to be done from the web interface now

closed 2019-03-04 by billz

aredey, 2024-01-09:
Did you get the resolution for this error? I have the same issue (with a fresh install on a raspi 2 using the 32-bit version of raspian with two USB wifi dongles), wlan1 (which should be the AP interface) never taking the fixed IP configured with the WebGUI on wlan1. Running this snippet fixing fixes the issue:

> ```
> #!/bin/bash
> inet=`hostname -I`
> if [[ $inet == 169*  ]];
> then
>     # no ip, we are in AP mode.
>     # set static ip 
>     ifconfig wlan0 down
>     ifconfig wlan0 192.168.1.1 netmask 255.255.255.0 up
>     sudo /etc/init.d/dnsmasq start
> fi
> ```

Thinking about doing this as well:
> /etc/rc.local or smth like this. let me know if it makes sense.

I've been scratching my head for days about what would be the "official" solution.
Thanks,