RaspAP/raspap-webgui #47: Forgotten Password Or Change Password

RyoisMC, 2016-11-28:
How would you change the password. In the raspap.php file it has a 'admin_pass' => line but if run it through a hash decrypter it fails. Thanks...

jrmhaig, 2016-11-30:
The default password is 'secret'. If, however, there is a file `raspap.auth` then the password in that file will be used instead. If that file exists then you can reset to the default by simply deleting it.
You may be able to brute-force the password but the hash is one way so you shouldn't be able to decrypt it.

RyoisMC, 2016-12-01:
@jrmhaig sorry I'm late but thx I'll try that now.

RyoisMC, 2016-12-01:
@jrmhaig where would that be located? /etc/raspap is empty

jrmhaig, 2016-12-01:
The `raspap.php` at least should be in `/etc/raspap`. Where is it now? And how did you install the software? I suspect that this is actually your problem.

By default the `raspap.auth` file does not exist but is created when you set the password.

billz, 2017-05-19:
Marking this closed. @RyoisMC please reopen if your issue is not resolved, thanks.

closed 2017-05-19 by billz