RaspAP/raspap-webgui #49: Please escape HTML entities in WiFi Nikname

RigacciOrg, 2016-12-17:
If the WiFi interface Nikname contains some HTML entities, the "Configure client" page (index.php?page=wpa_conf) get scrambled. My WiFi adapter has a nickname like this:

`wlan0     IEEE 802.11bg  ESSID:"Rigacci.Org"  Nickname:"<WIFI@REALTEK>"
`
and the page is badly composed:

`                  <input type="hidden" name="ssid0" value="Rigacci.Org"  Nickname:"<WIFI@REALTEK>" />
                  Rigacci.Org"  Nickname:"<WIFI@REALTEK>                </td>
                              <td><span class="label label-warning">X</span></td>`

RigacciOrg, 2016-12-18:
There are at least two issues: when preg_match() the ESSID from iwconfig output, you must get the first string enclosed in double quotes, (file configure_client.php, line 137):

    if (preg_match( '/ESSID:\"([^"]+)\"/i',$line,$iwconfig_ssid )) {

the you must at least use htmlentities($ssid) at line 176 of the same file.

billz, 2017-05-19:
Good catch, thanks.

commit 261c9047 (2017-05-19): Resolves #49
closed 2017-05-19 by billz