RaspAP/raspap-webgui #50: Feature request: support for more dnsmasq options

ScrumpyJack, 2016-12-19:
raspap currently crushes each config file when you save, so it would be nice to have a few more options with dnsmasq so that our extra setting don't vanish

#which address to listen on
listen-address=

#support for interface binding
bind-interfaces=

# to tell dhcp clients which dns host to use 
dhcp-option=6,8.8.8.8,8.8.6.6
#other nice to have options
dhcp-option=1 netmask
dhcp-option=3 router
dhcp-option=15 domain-name
dhcp-option=119 domain-search

# to disble dns server
port=0

#to enable dns server 
port=53
server=

#option to not load /etc/resolv.conf
no-resolv

#support for multiple interfaces
interface=br0
interface=wlan0
interface=wlan1
interface=eth0
interface=eth1
dhcp-range=br0,192.168.4.3,192.168.4.252,255.255.255.0,12h
dhcp-range=wlan0,192.168.5.3,192.168.5.252,255.255.255.0,12h
dhcp-range=wlan1,192.168.6.3,192.168.6.252,255.255.255.0,12h
dhcp-range=eth0192.168.7.3,192.168.7.252,255.255.255.0,12h
dhcp-range=eth1,192.168.8.3,192.168.8.252,255.255.255.0,12h

#support for creating infinite leases from the list of clients
dhcp-host=11:22:33:44:55:66,192.168.0.60,infinite

billz, 2017-05-19:
It's a nice idea to have more granular control over dnsmasq. I'll look into grouping these settings under an 'Advanced' tab in the DHCP panel.

EricPoggel, 2019-12-26:
I also needed my custom `dnsmasq.conf` values to not be overwritten, so I wrote a class that allows getting and setting values from a .conf file without modifying other values:

https://gist.github.com/EricPoggel/9caadfb445c0928196144aa489c97755

Here is a new version of the part of `/includes/dhcp.php` that uses this class to update values in `dnsmasq.conf` without erasing other values:

https://gist.github.com/EricPoggel/b4a1295350cc5ca788a5505446295abf

Code in `includes/hostapd.php` also writes to `dnsmasq.conf` and should be updated.  But I haven't touched it because I don't understand what the $wifiAPEnable switch is doing.

I release all this code into the public domain, so anyone may use it.  I just finished writing this so it may have bugs.  No warranty : )

billz, 2020-03-12:
Closing with #403 

Rather than provide a UI to manage these settings, this allows custom configs to be managed separately.

closed 2020-03-12 by billz