RaspAP/raspap-webgui #52: Copy SD to new Raspi 3

PureGrain, 2016-12-28:
Hello folks and good day. I have the software running on a Raspi 3 and it's working great, although when I make a copy of the SD and move it to another Raspi, the Wifi Does not want to come up for nothing. Any ideas?

RyoisMC, 2016-12-30:
Could it be that its trying to find a different WiFi card? Did you make an .img and burn it onto the SD card? If you can can you explain how you tried to migrate to another RPi?

PureGrain, 2016-12-30:
I started over with a fresh image and so long as I disconnect the WiFi from the client before copying the card, everything goes right. Now, on the other hand I'm having issues with the files copying correctly in the web directory.

I'm using ApplePi-Baker on MAC to create an image of the card and then dumping the image back on other cards. When I do this now, the wifi seems to be working out since changing what I mentioned above but, certain files are not copying to the image from the /html folder. It's at random as well when it picks files not to copy over. So when I fire up the image on another RasPi I have to copy those files manually in order for the web framework to work. It's like they are missing all together. Again, this happened a couple of times and it's just picking random files not to copy. (weird I know)

RyoisMC, 2016-12-30:
What device are you trying to connect to the new AP? Did it work on the older one?

PureGrain, 2016-12-30:
Everything connection wise is now working. I can use my iPhone, iMac and al of that works now and is copying over fine so long as I delete and remove the client before I copy the card. Hence taking it back to "local ip" since I'm using dhcp. The problem now is with the copying of files. No matter how I copy the card "dd" or Apple-Pi Baker it somehow missed important .js or .css files on the new card. These are at random as well when it misses them. I can go to the directory where they should be and they are just not there after the card copy.

jrmhaig, 2017-01-02:
Is the Pi you are copying to the same version as the old one? In particular, are they both Pi 3 or both not Pi 3? The only thing I can think of for your first problem is that the wireless device is different.
I have no idea why you might be missing files in the /html directory (I assume you mean the web root, which should be `/var/www/html` by default). If the "dd" is failing for some reason you are more likely to get a completely corrupted image rather than individual files missing. Are the two SD cards the same size? One possibility is that some of the files are at the end of the disk and so are getting missed off when copied onto a smaller disk but I would expect it to not work at all.

jrmhaig, 2017-01-02:
Also, I have never used ApplePi-Baker so I don't know exactly what it is doing. Is it possible that it is doing more than simple copying an image with "dd"? It looks like it creates images for use with NOOBS so maybe there is a more involved install process.

PureGrain, 2017-01-02:
Solved the issue but I will reply to give insight.

Yes, both are RasPi 3's

I was burning the images to an external USB 3.0 SATA Drive and I do believe that's what caused the problem. When I burn it to my local SSD the image does fine. So it's not anything to do with this application at all.

Thanks for the help folks!

closed 2017-01-02 by PureGrain