RaspAP/raspap-webgui #53: Hostapd Not Restarting

PureGrain, 2017-01-06:
Good afternoon folks. Thanks for all the help in advance. I've tried a few things and cannot seem to get this to work.

First thing is I set my /etc/dhcpcd.conf to comment out static IP settings because if I don't I do not get internet connection through my router after I setup as AP because it assigns me a local IP of 10.3.141.??.

If I configure the AP to my Router password it grabs an IP from my router and I can switch back to my network and log into the web interface again.

Here is where the problem is:

If I remove the password for the access point and just want to go back to local access it hangs. It does not restart the hostapd service. It's dropping the IP and everything but it's not refreshing and picking back up a local IP. I have to stop and restart the hostapd settings or reboot the system.

Any ideas?

jrmhaig, 2017-01-10:
Hello.

When you say you remove the password do you mean you remove it from the Raspap web interface?

If you disconnect from the router then I think it will drop the IP address and so, unless you put back the static settings in `/etc/dhcpd.conf`, it will not have any settings. Does this sound like what is happening in your case? This is a known issue (#46) but I'm afraid I haven't had much time to look at it.

PureGrain, 2017-01-10:
Yes, when I remove the password "delete button" from the web interface. Which I understood I lost the IP and therefor have no connection. I will write in some code to restart the hostapd app after the file is deleted to resolve the problem. Thanks for your time. If I get it working properly I'll provide the code here.

closed 2017-01-10 by PureGrain