RaspAP/raspap-webgui #54: Use isc-dhcp-server instead of dnsmasq

bizwoo, 2017-01-17:
I  had a working  raspberry pi AP previously configured using isc-dhcp-server. I installed RaspAP which I think is an awesome webgui to easily controlling the AP. However, it looks like it cannot coexist with  isc-dhcp-server and uses dnsmasq to handling the assignment of the IP addresses, DNS, etc to the connected clients. Is it possible to get RaspAP to use isc-dhcp-server instead of dnsmasq. The config file for isc-dhcp-server  is located at /etc/dhcp/dhcpd.conf
Thank you.

jrmhaig, 2017-01-27:
I think that to use an alternative to dnsmasq you would need to make some changes to the code, in particular `includes/dhcp.php`. The `dnsmasq` (and other) command line tools are called using the `exec` method so these are the things to look for.

If you do feel inclined to create a pull request for this then it would be helpful if it could be configurable via the `raspap.php` configuration file.

billz, 2019-11-05:
The administration interface for isc-dhcp-server is different enough from dnsmasq that this would necessitate significant code changes. Given that dnsmasq does everything needed for the project, this is not likely to happen.

I would, however, encourage you to create a PR or fork the project.