RaspAP/raspap-webgui #56: Client UI only

nooryani84, 2017-01-28:
Sorry for posting this in the wrong place... If I only want to use the client-section of your GUI for connecting my RPi to the internet, what would I have to comment out/delete?

jrmhaig, 2017-02-07:
Sorry for being slow in replying. You need to comment out or delete the last four lines of `/etc/dhcpcd.conf`, which look like this:

```
# RaspAP-WebGui wireless configuration
interface wlan0
static ip_address=10.3.141.1/24
static routers=10.3.141.1
static domain_name_server=8.8.8.8 8.8.4.4
```

This is a known bug (#46) but I haven't had a chance to look at it.

nooryani84, 2017-02-07:
Will this remove all other sections of the GUI?   I've been trying to look at the code and remove any references to features besides the client options.   I've removed the following files and references to them in raspap-webgui/index.php:

dhcp.php
functions.php
hostapd.php

My goal is to trim all features and files that don't have to do with the Client, essentially just removing Hotspot and DHCP.   I may have to learn more in order to do this effecitvely, though I really like the way you have set things up here in general :)

jrmhaig, 2017-02-07:
Ah, I understand. You want to remove them from the UI. Have a look at this to edit the navigation menu:

* https://github.com/billz/raspap-webgui/blob/master/index.php#L123-L152

and this to prevent them being accessible directly from a url:

* https://github.com/billz/raspap-webgui/blob/master/index.php#L170-L200

Does this help?

nooryani84, 2017-02-07:
Yes thank you :)

closed 2017-02-07 by nooryani84