RaspAP/raspap-webgui #61: Pi zero

Arky92, 2017-03-22:
After i did the automatic install and i rebooted, the pi won't to connect to WiFi or over USB to the PC with otg mode enabled.
I'm an ower of the new pi zero w, my modem range ip is "192.168.1.2" to "192.168.1.250"

jrmhaig, 2017-03-22:
Sorry, you may have hit this bug: #46 

Can you please try editing `/etc/dhcpcd.conf` and removing the lines:

```
static ip_address=10.3.141.1/24
static routers=10.3.141.1
static domain_name_server=8.8.8.8 8.8.4.4
```

Arky92, 2017-03-22:
Thanks for the quick answer! It works!

closed 2017-03-22 by Arky92

Arunms123, 2017-09-05:
I'm using RaspAP and Currently my Pi zero is connected to WiFi, but when i configure hotspot and start it is showing an error as "HostAPD is not running"