RaspAP/raspap-webgui #67: Eth0 Static IP Configuration

robertsLando, 2017-04-06:
I think this is a must-have tool for every Raspberry PI. In my opinion the only thing missing is a configuration page for the eth0, it would be nice to be able to set the static IP of eth0 from this interface. What about this feature in a future release? Just need to edit the file /etc/dhcpcd.conf to configure eth0 static IP.
Let me know.

Thanks

jrmhaig, 2017-04-11:
Thanks. This would be a useful feature and I'll give it some thought.

Unfortunately, development of new features happens when I get time and I haven't had much recently.

billz, 2017-06-06:
There's a related request #50 for more granular control over dnsmasq settings. I'll look into combining these into a future update.

Mark-81, 2017-08-26:
Any news about the ethernet configuration? I would fork the project if you don't plan to release the update soon. Thanks for the great job anyway!

SirLagz, 2017-10-11:
@Mark-81 Feel free to fork it and submit a pull request to merge it back into this main branch.

SirLagz, 2017-10-30:
Pull Request #131 addresses this issue

billz, 2017-11-17:
@robertsLando @Mark-81 the new Configure Networking interface should allow you to do this. Closing this issue for now. Please reopen if needed, thanks.

closed 2017-11-17 by billz