RaspAP/raspap-webgui #68: include modem 3g/4g as source for internet

mengelh, 2017-04-08:
It would be great if we were able to use and configure 3g/4g modems as source for internet, using the web UI.

May be using "wvdial" for the configuration and "usb-modeswitch" for the mode switch of the 3g/4g stick.

jrmhaig, 2017-04-11:
I don't have one of these but if someone wants to make a pull request it could certainly be considered.

zbourdeau, 2017-08-04:
I have actually coded some of this functionality in. I am using the PiAnywhere 4G hat for a modem. What I did was create a new dashboard based on the wifi dashboard and I display some of the statistics of the connection on that page as well as have up/down interface functionality. I haven't quite figured out the routing yet to switch between the WLAN and PPP connection seamlessly which I would like to get working. I'll try to PR this in when I get some free time.

jensjakob, 2018-04-13:
I just put my 4G dongle in the USB-port and installed RaspAP. It worked.

hallbrad, 2018-04-15:
@jensjakob what 4G dongle are you using? Could you share a link to it? Thanks!

jensjakob, 2018-04-15:
@hallbrad Huawei E3372. It's branded from Tele2, but they use another brand today. No link.

jtsmedley, 2019-11-22:
Need any help on this?

Would be extremely helpful for my project

billz, 2019-11-23:
@jtsmedley PRs all always welcome ;)

jesgeerts, 2019-11-24:
An RNDIS 3g/4g modem through USB will work. Tested it with a TP LINK 7350.

Dude4Linux, 2020-01-22:
I'm using a Huawei E397u-53 USB 4G LTE modem (Cricket Brand) with the T-Mobile Network.
I found it easiest to manage by installing usb-modeswitch modemmanager and network-manager.
A connection has to be established from the command line, but it only needs to be done once.
```
nmcli connection add type gsm ifname 'cdc-wdm0' con-name LTE apn 'Fast.t-mobile.com'
```
Still working out how to switch between WiFi and LTE.

olewsaa, 2020-01-29:
For my use case [access point onboard a boat](https://sites.google.com/site/olewsaa/yacht-server/raspberry-pi-as-a-router-gateway) I would like to use a USB 4G(5G) dongle as the RPi is inside en the wifi USB long range antenna and the USB 4G dongle need to be outside. As most USB 4G dongles are not waterproof they need to be in box when outside. This is the trivial part. 

The part that require programming and algorithm is when to change from wifi to 4G. Doing it seamlessly might not be a good idea as usage when connected to wifi is more or less unlimited (limited by bandwidth) while 4G access is limited by amount of data (data package). Forcing the users to open the web-gui and select 4G when wifi is not present of go offline is one option. Limiting bandwidth in some way another. Family members and crew watch TV and Netflix when on wifi, which they cannot do over 4G (assuming a limited data package, which most of us have).
There are a lot of thinking to be done to come up with smart way of dealing with this.

zbourdeau, 2020-01-29:
olewsaa, That is my exact use case and I am curious to know more about how your project is working for you and possibly collaborating with you on this. I am using a USB Wifi dongle (with external antenna) and the Six Fab LTE hat https://sixfab.com/product/raspberry-pi-base-hat-3g-4g-lte-minipcie-cards/ with an external antenna. I 100% agree that its a challenge to do the transition from WIFI to Cellular and back and have not solved this problem yet. When I was running Jessie with the Altitude tech LTE card (which I recently abandoned due to the physical on/off switch and lack of support) using just the routing table I was able to get the wifi to automatically switch to cellular once but i would have to restart/ manually change the routing table to go back to Wifi.  Have you looked at this project: https://github.com/drsound/fault_tolerant_router ? This seems like it may be a possible solution.

sibero80, 2020-03-24:
I have been able to get a Huawei E303 working with RaspAP, but I find the approach cumbersome and very improve-able.

I'm by no means a developer, but I see that this could be done very elegantly by someone more skilled than me.

Here's my approach:
(Followed [this ](https://www.raspberry-pi-geek.com/Archive/2016/15/Creating-a-3G-hotspot-with-the-Raspberry-Pi)and [this ](https://gist.github.com/toke/52a10c6c59d0f1df2b4c)to set up ppp0)

- Installed ppp wvdial 
- ran wvdialconf to generate wvdial.conf
- modified wvdial.conf to include APN settings
- created ppp0 in etc/network/interfaces.d/ppp0
- created startup wait script
- modified default route following the answer to [this](https://raspberrypi.stackexchange.com/questions/15119/force-raspberry-to-get-internet-from-specific-network)

And voilá, I get my portable 3G hotspot.

Desired funcionality:
- Include pp wvdial packages
- User interface to enable 3g stick (runs wvdialconf in background)
- User interface to add APN (wvdial.conf), Phone #, User and Password. The only thing required by most carriers today.
- Advanced user interface to edit wvdial.conf parameters.
- Swith to enable/disable internet from 3G (brings ppp0 up and toogles default route, or more elegant solution)

cedpfi, 2020-04-08:
Hello,
I have got a raspberry pi 3 and I connected a USB modem 4g. This modem run with network manager.
Now I would like to do a hotspot with raspap and share the 4g connection with the devices connected with my raspberry but when my hotspot is actived, my internet connexion with my dongle modem doesn't work. 
Have you got an idea to resolve this problem?

lakeconstance78, 2020-05-27:
Hello,

I tried a lot to get 3G + RaspAP working. 

The following steps are working with my Raspberry 1+2, Huawei E160 3G-Dongle + Edimax WLAN-USB-Stick.

1. Install Raspian + Update + Upgrade
2. Install RaspAP
3. Plugin 3G-Dongle
4. Install wvdial -> `sudo apt-get install wvdial.`
5. `sudo nano /etc/wvdial.conf` and insert the following. Then put in the correct APN, user and password.
```
[Dialer Defaults]
Init1 = ATZ
Init2 = ATQ0 V1 E1 S0=0 &C1 &D2 +FCLASS=0
Init3 = AT+CGDCONT=1,"IP","PUTAPNHERE","0.0.0.0",0,0
Modem Type = USB Modem
Baud = 1000000000
New PPPD = yes
Modem = /dev/ttyUSB0
ISDN = 0
Phone = *99***1#
Password = PUTINHERE
Username = PUTINHERE
Stupid Mode = 1
Dial Attempts = 2
Carrier Check = no
```
6. Start wvdial with `sudo wvdial`. The connection will be established.
7. If you do a `ping google.com` it should work.
8. To route the internet traffic from the 3G-Dongle to the RaspAP just `sudo nano /etc/ppp/peers/wvdial ` and add:
```
noauth
name wvdial
usepeerdns
defaultroute
replacedefaultroute
```
9. To start the 3G connection on every startup `sudo nano /etc/rc.local` and add
    `wvdial` before the `exit 0`.
10. Reboot your Pie.

This is the easiest solution (but quick and dirty) and you not have to change the routes or somewhat manually.

But be careful: All ports will be open. So you have to config the firewall...

Regards,

Ben.

billz, 2021-05-01:
Thanks to @zbchristian's work this has now been implemented. Details at https://docs.raspap.com/net-devices/

closed 2021-05-01 by billz