RaspAP/raspap-webgui #69: Can be used as a client for setting wifi bridge instead of hotspot

dkebler, 2017-04-15:
Does the "client" page link in your screenshot mean that this webgui can be used to set the WPA key/pw for the adaptor to act as a simple client to another WAP?

I am setting up a wifi bridge (internet=>wifi=>ethernet) as opposed to a hotspot router and just need to be a able to set the WPA passwords conveniently instead of having the VNC/SSH to the RPi.  This is going in my travel trailer to it will be connecting frequently to different WAPs.

If so should I just do the manual install?

zlolz, 2017-04-16:
"Yes", kinda.  You can wrangle it to support the 'travelling/security hotspot' case with workarounds, but not out of the box and it's not a simple thing to fix architecturally.  Issues #20 and #21 get into that a bit (indirectly).

But what you want *is* do-able - I do it both wired and wireless, with roaming - and you *can* use this project as a base to expand on.  If you let webgui handle the initial ap setup only, and have something like wpa-roam do the travelling/roaming side you will be 90% there (at which point you realize you don't need to use webgui any more)...

dkebler, 2017-04-16:
I did go ahead and install it and now see that it doesn't use iwlist or other tool to find available hotspots, it just looks in the supplicant file for what's already there.    Plus I'd have to undo all the wap settings for static ip and dhcp server....so you are right @zlolz this isn't the project to work from..  I found this one  https://github.com/piotrmaslanka/supplicanted but I'm a nodejs coder.  I'm just going to vnc for now.  Maybe write a web interface using nodejs later.

closed 2017-04-16 by dkebler

zlolz, 2017-04-16:
> I did go ahead and install it and now see that it doesn't use iwlist or other tool to find available hotspots, it just looks in the supplicant file for what's already there.

I don't believe that's the case.  It uses wpa_cli... (I think?  I no longer use this project)

> so you are right @zlolz this isn't the project to work from.. 

That's not what I said.  Sounds like this project doesn't do what you want out of the box, but webgui can help with the ap side to start with (if you are sold on needing a gui - nothing wrong with that).

FWIW:  I also hoped (from the screenshots) this would solve the 'travel/security' case.  Seemed logical.  But it didn't.  I hung around for a while and maintained a patchset while trying to wrangle it into doing so, but eventually realised this was for a different audience and I would be solving a problem most folks didn't have, so I binned the patches and re-wrote as a combined server/gui/control panel in golang which was as dreadful as it sounds then further simplified (in the field) to a couple of scripts and configs that I didn't have to think about.  I imagine you are going down the same route.  Everyone is different but I roamed around asia and the us for months without ever needing to select an ap via gui - wpa_roam is quite flexible/'programmable' and hostapd pairs nicely.  Add transparent vpn(s) and you have a nice travel box for all your devices.

Might want to keep the issue open for an official response - I'm just some interfering guy.