RaspAP/raspap-webgui #71: Client settings - How to manually switch from one wifi to another

magneticlab-ch, 2017-05-09:
Hi guys , 1st of all thank you for this great job !
I actualy use my Rpi3 build in wifi connection in adhoc mode (as an access point or a repeater if you want) meaning that I can join my Rpi wifi network with my phone for example.
Than , when i go to Raspap webpage under client settings,
 i would like to enter a password for one or another available network and activate it.

The problem is that when i click on update , the wifi wont come up anymore . it just stops.
I can I change this behaviour to get my wifi stating up again on the one i choose ?

magneticlab-ch, 2017-05-09:
ok, the problem is the file **/etc/hostapd/hostapd.conf**
which is not updated as I want.

billz, 2019-04-13:
Fixed with #297

closed 2019-04-13 by billz