RaspAP/raspap-webgui #72: Unit Lighttpd.service entered failed state

ikbelkirasan, 2017-05-11:
Hello,

Using a fresh installation of Raspbian Jessie Lite and the quick installer script, everything went well. After rebooting, I connected to the new access point I tried to access the web UI. The connection was refused because Lighttpd failed.

Running `sudo service lighttpd status` gives the following:
```
● lighttpd.service - Lighttpd Daemon
   Loaded: loaded (/lib/systemd/system/lighttpd.service; enabled)
   Active: failed (Result: exit-code) since Thu 2017-05-11 16:37:39 UTC; 2s ago
  Process: 2785 ExecStart=/usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf (code=exited, status=255)
  Process: 2775 ExecStartPre=/usr/sbin/lighttpd -t -f /etc/lighttpd/lighttpd.conf (code=exited, status=0/SUCCESS)
 Main PID: 2785 (code=exited, status=255)

May 11 16:37:39 raspberrypi lighttpd[2775]: Syntax OK
May 11 16:37:39 raspberrypi systemd[1]: Started Lighttpd Daemon.
May 11 16:37:39 raspberrypi systemd[1]: lighttpd.service: main process exited, code=exited, status=255/n/a
May 11 16:37:39 raspberrypi systemd[1]: Unit lighttpd.service entered failed state.
```
The folder `/etc/lighttpd/conf-enabled` contains two configuration. if I remove either one of them and run `sudo service lighttpd restart`, then when I try to access the web page again, it responds with 403 - Forbidden error.

What could be the problem?

Thanks!

billz, 2017-05-14:
Normally there are two default configuration files in `/etc/lighttpd/conf-enabled` that support fastcgi. The RaspAP installer doesn't touch this folder, but lighttpd needs them. Start by moving those files back or reinstall lighttpd (`sudo apt-get install lighttpd`) then try accessing the web UI again. If it still fails, the error log may have some clues: 

`sudo cat /var/log/lighttpd/error.log` 

Just to be sure nothing strange was written to the lighttpd config, you can also perform this check:

`lighttpd -t -f /etc/lighttpd/lighttpd.conf`

billz, 2019-04-13:
Closing due to inactivity.

closed 2019-04-13 by billz