RaspAP/raspap-webgui #74: Add OpenVPN support

billz, 2017-05-20:
[PiVPN](https://github.com/pivpn/pivpn) is a robust OpenVPN project specifically for the RPi. Stub support for OpenVPN exists in RaspAP. This could potentially be extended to manage PiVPN from RaspAP (add/list/revoke client ovpn profiles) with the `pivpn` command. 

A simple way to implement this is to write a value to `/etc/raspap/raspap.conf` if the user opts to configure PiVPN with the Quick Install (see #73) and use this to enable the OpenVPN panel in the web interface.

MrSinaRJ, 2017-09-09:
Hi, I have just found this beautiful job you have done.
Is there anyway we get openVPN soon?

billz, 2019-11-16:
@MrSinaRJ see #147. closing this as there is more detail in the other referenced issue.

closed 2019-11-16 by billz