RaspAP/raspap-webgui #77: rPI Zero W needs AP on same channel as client wifi

andyjbm, 2017-05-24:
It's possible to use a pi Zero W as an AP/Client using the on board wifi nic as both AP wlan and client wlan.

The setup is reasonable straightfoward ;-) requiring a few tweeks shown below but the reason for this issue is the problem that in this config (using onboard wifi for both AP and client) the wifi nic requires that the AP be on the same channel as the client it has connected to.

So a mod is needed to change the AP channel (in hostapd.conf) to that of the client when it connects to a client ssid... otherwise the whole wifi client and AP falls over.

So example tweeks as promised to get the shared nic setup working in the 1st place.
This uses wlan0 as the client facing interface and creates uap0 as the AP interface.
Yes I believe it only works this way round.

This in /etc/rc.local:
```
#Set up IP forwarding and NAT routing
sudo echo 1 > /proc/sys/net/ipv4/ip_forward
sudo iptables -t nat -A POSTROUTING -j MASQUERADE

#Create the AP Interface:
sudo iw dev wlan0 interface add uap0 type __ap
sudo ip addr add 192.168.12.1/24 dev uap0

sudo service hostapd restart
sudo service dnsmasq restart
```

Here's an example /etc/network/interfaces:
'''auto lo
auto eth0
auto wlan0
auto wlan1

auto uap0

iface lo inet loopback
iface eth0 inet dhcp

allow-hotplug wlan0
allow-hotplug wlan1

iface wlan0 inet dhcp
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

iface uap0 inet static
  address 192.168.12.1
  netmask 255.255.255.0
  network 192.168.12.0
  broadcast 192.168.12.255
  gateway 192.168.12.1
'''
and finally an example /etc/hostapd/hostapd.conf:
'''
# ctrl_interface=/var/run/hostapd
# ctrl_interface_group=0
beacon_int=100
auth_algs=1
wpa_key_mgmt=WPA-PSK
ssid=raspi-webgui
channel=6
hw_mode=g
wpa_passphrase=ChangeMe
interface=uap0
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
country_code=GB
hw_mode=g
'''

jspraul, 2017-09-28:
https://albeec13.github.io/2017/09/26/raspberry-pi-zero-w-simultaneous-ap-and-managed-mode-wifi/

https://news.ycombinator.com/item?id=15351778

SirLagz, 2017-10-03:
That also need a new virtuasl interface created for the AP or Client connection.
RaspAP currently can't handle more than one WiFi connection, but I'll put this onto our ToDo list.

billz, 2019-04-19:
@andyjbm this has been implemented with #297. Feedback is welcome, thanks.

andyjbm, 2019-04-20:
@billz ok cool! Thanks for your work and the heads up. My head's in another project for the next month or so but I'll defo take a look as soon as I'm back there.
Many thanks.

francescoatmitie, 2019-06-27:
I'm currently using a Rasp PI3+ with latest RaspAP release and it looks like I have the same problem, the AP channel needs to match the WiFI client one in order to be able to connect to AP.

Is there any clear workaround or configuration that I missed to avoid this issue?

billz, 2019-06-27:
@francescoatmitie the channel used by the virtual interface (uap0) with a managed mode AP must match the channel that wlan0 uses for a WiFi connection. this is a limitation of the adapter, there is no workaround.

closed 2019-06-27 by billz

francescoatmitie, 2019-06-27:
@billz thank you for the clarification. And well done to put together this useful application.

francescoatmitie, 2019-06-27:
@billz maybe a naive question, but is there any way to automatically detect the WiFi Client channel and change AP side automatically to match it?