RaspAP/raspap-webgui #78: Can't seem to get this working with a RPi3 + USB WiFi for Client side

jbishop129, 2017-06-02:
Hey guys, sorry I know this isn't a 'support forum', but seemed like the only place I could turn to. Vanilla install of Rasbian Jesse on a RPi3, supported USB WiFi Dongle, ran the one-line quick installer from the readme and it works just fine. I want to use the built-in RPi's WiFi (wlan0) as the Hotspot side, and the USB WiFI (wlan1) as the Client side. I can connect to the Hotspot side just fine with phones and laptops, can hit the configuration webUI just fine. I can get the Client interface to connect to a WiFi network just fine, I see it pulls an IP (ifconfig) and is in fact attached (iwconfig) to the network, but it doesnt actually let clients attached to the Hotspot connect outbound through that connection. If I connect Ethernet hard-wired to eth0, it works like a champ, just doesnt seem to bridge / NAT from wlan0 -> wlan1 as far as I can tell. Sort of seems like it's hard-coded to wlan0 -> eth0 only.  I feel like I'm missing something very simple! Any pointers? Thanks! --Joe

billz, 2017-06-03:
> I want to use the built-in RPi's WiFi (wlan0) as the Hotspot side, and the USB WiFI (wlan1) as the Client side

@jbishop129 unfortunately, I don't have an RPi3 or two WiFi dongles to test this case. What you describe should be doable, though. A routed+NAT setup between the two wireless interfaces is what you're after. RaspAP doesn't currently support this type of configuration in the UI, but it might be a candidate for an advanced option in the future. 

Again, I'm unable to verify on my hardware, but this is how to configure NAT between wlan0 (the AP) and wlan1 (the client):

```
    echo 1 > /proc/sys/net/ipv4/ip_forward
    iptables -A FORWARD -i wlan0 -o wlan1 -j ACCEPT
    iptables -A FORWARD -i wlan1 -o wlan0 -m state --state ESTABLISHED,RELATED -j ACCEPT
    iptables -t nat -A POSTROUTING -o wlan1 -j MASQUERADE
```

You may want to list your current iptables rules (`sudo iptables -S`) and optionally flush any rules that conflict with the above.

jbishop129, 2017-06-03:
Hi @billz thanks for replying to me. I did end up getting this working; it turned out my client-side wlan1 interface was set to ```iface wlan1 inet manual```, and needed to be ```iface wlan1 inet dhcp``` in ```/etc/network/interfaces``` so it would get a full DHCP configuration from the client network. 

I may be slightly confused: Is the implementation intention with RaspAP to use a single wlan0 interface for both the hotspot side, and the client side?

billz, 2017-06-06:
Glad you got it working. Initially it was intended more as an either/or use, as many Pi owners simply wanted an easy way to manage client connectivity. Nowadays, RPi Zeros are being configured to operate in an AP and client mode on the same wlan0 interface. With the RPi3's onboard wifi it makes even more sense to support multiple wlan interfaces in RaspAP. 

Thanks for your contribution. We'll look to integrate this soon.

anthortic, 2017-06-11:
Hello, I also know this isnt a support forum but I have finished on a solution that is functional for me and ahould help jbishop129.

If its helpful to anyone then I figured I'd best post close to source. I hope its a useful contribution!
 
I am using an rpi3 and raspap-webgui with two wifi adapters (On board chip (AP) and usb RTL8191SU (Client)).

The client won't be fixed to one AP because it is installed in my campervan and the usb adapter has an external sma connection so I can hook up a substantial aerial a distance away from the pi. For this reason a router setup is more suitable than a bridge because it will be the home network for our devices - therefore my iptables are different to billz suggestion.

For this reason I have setup my iptables and iprouting scripts through cron. Setting the ip at boot is not flexible enough when I am travelling and when using a public hotspot I only want the hassle of registering one device for all the others and have iptables respond automatically - so the ui is really great!!! thank you :)

After the raspap-webgui quick installer I setup the scripts in this way:

wlan0 - access point
wlan1 - client

1. resetiptables - to flush all rules out of the system
`#!/bin/sh
IPTABLES="$(which iptables)"
# RESET DEFAULT POLICIES
$IPTABLES -P INPUT ACCEPT
$IPTABLES -P FORWARD ACCEPT
$IPTABLES -P OUTPUT ACCEPT
$IPTABLES -t nat -P PREROUTING ACCEPT
$IPTABLES -t nat -P POSTROUTING ACCEPT
$IPTABLES -t nat -P OUTPUT ACCEPT
$IPTABLES -t mangle -P PREROUTING ACCEPT
$IPTABLES -t mangle -P OUTPUT ACCEPT
# FLUSH ALL RULES, ERASE NON-DEFAULT CHAINS
$IPTABLES -F
$IPTABLES -X
$IPTABLES -t nat -F
$IPTABLES -t nat -X
$IPTABLES -t mangle -F
$IPTABLES -t mangle -X`

2. setupiptables - that parse the client ip before setting rules
`#!/bin/sh
 IPT=/sbin/iptables
 LOCAL_IFACE=wlan0
 INET_IFACE=wlan1
 INET_ADDRESS=$(/sbin/ip -o -4 addr list wlan1 | awk '{print $4}' | cut -d/ -f1)
# Flush the tables
 $IPT -F INPUT
 $IPT -F OUTPUT
 $IPT -F FORWARD
$IPT -t nat -P PREROUTING ACCEPT
 $IPT -t nat -P POSTROUTING ACCEPT
 $IPT -t nat -P OUTPUT ACCEPT
# Allow forwarding packets:
 $IPT -A FORWARD -p ALL -i $LOCAL_IFACE -o $INET_IFACE -j ACCEPT
 $IPT -A FORWARD -i $INET_IFACE -o $LOCAL_IFACE -m state --state ESTABLISHED,RELATED -j ACCEPT
# Packet masquerading
 #$IPT -t nat -A POSTROUTING -o $LOCAL_IFACE -j MASQUERADE
 $IPT -t nat -A POSTROUTING -o $INET_IFACE -j SNAT --to-source $INET_ADDRESS`

3. setuproutes - to enforce that the client is the only gateway
`#!/bin/sh
#/etc/init.d/setuproutes
#
### BEGIN INIT INFO
# Provides: default routes for wlan1
# Required-Start: $syslog $network
# Required-Stop: $syslog
# Should-Start:
# Should-Stop:
# X-Start-Before:
# X-Stop-After:
# Default-Start: 2 3 4 5
# Default-Stop: 0 1 6
# X-Interactive: false
[... 199 lines cut ...]
    } else {
      echo 'wlan1 Interface already down';
    }
  } elseif( isset($_POST['ifup_wlan1']) ) {
    exec( 'ifconfig wlan1 | grep -i running | wc -l',$test1 );
    if($test1[0] == 0) {
      exec( 'sudo ifup wlan1',$return1 );
    } else {
      echo 'wlan1 Interface already up';
    }
  }
  ?>`

Thats all I did I'm sure, just need to play with incorporating pi-hole which sits in a subdirectory of /var/www/html/admin so will sit alongside raspap-webgui with no conflicts????? Not sure...

Thank you very much @billzimmerman, @jrmhaig and @SirLagz I enjoyed learning something new and look forward to simplfied internet access while on the road through Europe.

Cheers,
@antsinflowers

sources -

1.,2.,3. - (Slightly modified from) https://rbnrpi.wordpress.com/project-list/wifi-to-ethernet-adapter-for-an-ethernet-ready-tv/
4. - Stackoverflow, or the like
5. - https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=55527

randyoo, 2017-06-25:
Came here to say "thanks" to @anthortic , as this is the exact scenario I wish to use my RasPi in. Also thanks to the authors of raspap! @billz , it certainly would be nice to have this implemented in the GUI, but thanks for sharing the fruits of all your efforts thus far!

billz, 2017-06-26:
@anthortic this is a super valuable contribution. Clearly this is a use case a growing number of RPi and RaspAP users are seeking. I've bumped this up on my list of priorities. Many thanks.

MBishton, 2018-02-09:
I have been searching for days to find instructions that I, a new person, can follow, to do the exact thing that jbishop129 wants to do. Either with a dongle on a Pi3 or on a Pi0W as billz outlined. But I can't find any site that give it to me step by step. You all know what you are doing. Can someone point me to a URL that can step me through this? Many thanks!

jbishop129, 2018-02-09:
Holy cow, what a coincidence! I had been sidetracked on other projects, and
just yesterday while on vacation was looking at RaspAP's progress. I saw
this commit below that I'm excited to try when I get home, which looks like
it takes the hardcoded wlan0 interface and allows it to be modded easily. I
don't think the commit has been approved and merged yet, but certainly the
first thing I will be testing the I get home next week

https://github.com/billz/raspap-webgui/pull/140/files

Joe

billz, 2018-02-11:
@jbishop129 I tested and merged @emmanuelgeoffray's pull request. The ability to set the wireless interface from a global is indeed very useful.

jbishop129, 2018-02-11:
@billz thank you sir. I will pull and test this on my end shortly!

closed 2019-04-13 by billz