RaspAP/raspap-webgui #79: Bower?

jrmhaig, 2017-06-02:
@billz 

Quick question. Do you use Bower for package management or are all the files and directories a carry-over from where you copied the theme in the first place? I'm considering a clean-up but thought I would check first.

billz, 2017-06-02:
@jrmhaig Bower is indeed for package management and came across from the SB Admin 2 theme. 

Excluding `bower_components` will reduce the size of the repo, however there are different opinions on the [best way of handling these dependencies](https://addyosmani.com/blog/checking-in-front-end-dependencies/). I've thought of excluding this folder and making `bower install` part of the the Quick Install process. This way, only `bower.json` is needed and everything stays current. The potential downside is a dependency change breaking the app or network issues preventing things from loading correctly. 

There are pros and cons to both approaches. I opted to preserve the directory but am open to ideas.

jrmhaig, 2017-06-03:
Thanks. There appear to be a lot of unnecessary CSS and Javascript that is being pulled in as a dependency of the theme so I am thinking about cleaning it up. For example, [metis](http://mm.onokumus.com/) and [morris](http://morrisjs.github.io/morris.js/) appear to be quite powerful libraries but are not actually used.

When I tried installing Bower I got the warning that it is deprecated and Yarn should be used instead. What do you think about this? On the other hand, if I am able to recreate what SB Admin 2 gives us in plain Bootstrap it may be possible just to have the bootstrap css and js files plus a file or two for our own customisation and then Bower/Yarn would be overkill.

Also, from what I have read Bower/Yarn are intended for Javascript applications while RaspAP is mainly PHP and has minimal Javascript.

I'll see what I can do in a branch.

jrmhaig, 2017-06-03:
I have created a branch here:

  https://github.com/billz/raspap-webgui/tree/restyle

Most of the changes are removing anything Bower related. I have then put Bootstrap into `css/`, `js/` and `fonts/` as well as putting JQuery, which is a dependency of Bootstrap, into `js/`. The `index.php` has then been modified to use this instead of SB Admin 2.

It does look a bit different. In particular, I didn't realise that SB Admin 2 had changed the default colours, but these can be tweaked.

Thoughts?

billz, 2017-06-03:
Looks good to me. Much cleaner and it preserves Bootstrap, which is really the core. I cleared my cache and didn't notice any obvious changes in the UI.

If/when fancy menus and graphs are needed, we can look at reintroducing things like Metis and Morris. For now I'd say this a nice improvement. 

Would you like to put this into a pull request?

billz, 2017-06-03:
Ah, I see what you mean about the colors now. Did you include /dist/css/custom.css  + sb-admin-2.css? The Auth and System menus don't seem to load correctly 🤔

jrmhaig, 2017-06-05:
I'm holding off creating a pull request because I haven't actually tried it on a Pi yet so I haven't checked that it is OK. You appear to have already found a couple of broken links.

Also, I'll have a look to see if I can get the colours back. I deliberately cut it back to just the basic Bootstrap but haven't got round to reintroducing the styling from sb-admin-2 that we actually want.

billz, 2017-08-31:
@jrmhaig closing this for now. Feel free to reopen the issue so we can track progress. Thanks.

closed 2017-08-31 by billz