RaspAP/raspap-webgui #80: After admin login, interface is blank on raspberry pi 2 b+

guilhermemauro, 2017-06-06:
I intalled rasp-webgui with Quick Installer. My raspberry reboots and the network "raspi-gui" is created. A can connect in, surf on the web, but after I access http://10.3.141.1/ and login, the interface stay blank. No error occurs. Can anyone help me?

billz, 2017-06-06:
@guilhermemauro `password_verify()` is a native PHP function as of version 5.5. This error indicates that your PHP install is outdated. Also, your webroot appears to be `/var/www` while in Jessie `/var/www/html` is the default. 

Check your installed PHP version with `php -v`
and your Raspbian version with `cat /etc/debian_version`.

guilhermemauro, 2017-06-06:
Hi @billz thanks for helping. You're right. The Quick Installer, does not installed php. So, I manually installed php-5.6.30 and now I can see the interface, but the php code not is interpreted, look how it's shows: http://imgur.com/Olhr2D3 There is no error on lighttpd log.

billz, 2017-06-06:
Try giving the web service a kick? `sudo service lighttpd restart`

billz, 2017-06-12:
Closing this for now. If the issue persists please reopen it, thanks.

closed 2017-06-12 by billz