RaspAP/raspap-webgui #83: Disable RaspAP on boot

robertsLando, 2017-06-20:
I install RaspAP in every Raspberry I use but sometimes I don't need the wifi on so I would like to disable it from boot. How can I do this?

billz, 2017-08-31:
@robertsLando the best way to achieve this is to disable loading of the Broadcom wireless drivers by editing `/etc/modprobe.d/raspi-blacklist.conf`:

```
#wifi
blacklist brcmfmac
blacklist brcmutil
```

billz, 2017-09-11:
Closing this issue as there's been no activity. Reopen if needed, thanks.

closed 2017-09-11 by billz

noxware, 2022-10-16:
This worked for me:

```
sudo systemctl disable hostapd dnsmasq raspapd
sudo reboot
```

If you don't disable the `raspapd`, it will reconfigure the other two on boot.

After reboot, you can check that the `hostapd` is kept disabled with `sudo systemctl status hostapd`.