RaspAP/raspap-webgui #84: Temporary fix for openVPN -client

markdiesel, 2017-06-23:
Hi, I am new to this. So maybe not the correct way to add info.Excuse me if I am wrong ;-) 

Installed RaspAP on my Raspberry 3.  Did a manual install of openvpn (no pivpn) . 
enabled VPN in index.php, found client not working. Fixed it like this; 

1. add www-data ALL=(ALL) /etc/init.d/openvpn start to sudoers
2. add www-data ALL=(ALL)/etc/init.d/openvpn stop to sudoers
3.  find  in /var/www/html/include/functions.php  lin 247 : ($hostapdstatus[0] == 0)
4.  change to ($openvpnstatus[0] == 0  ) to show stop [Stop OpenVPN] in page (should look to status of vpn instead of hostapd 

vpn is working now. Although I need to refresh the page to see the effect of the start/stop. 
Did not test the server functionality yet.

MrSinaRJ, 2017-09-09:
Hi
Can you please explain it a little bit easier
I'm a total noob on this matter. step 1 and 2 are a little bit hard to understand and do.
I'll be glad to do it manually before it get fixed permanently

billz, 2019-11-16:
@markdiesel OpenVPN client configuration support is now in beta. See #147. please report any issues there, thanks

closed 2019-11-16 by billz