RaspAP/raspap-webgui #85: Very slow and errors in log file

rovincent, 2017-07-04:
Hi Guys,

Everything is installed i believe and the latest versions. using Rasbian Jessy. followed the quick install but got error because hostap and dnsmasq were already installed, ran command as mentioned. Made sure all manual steps were followed.

Wifi AP comes up and i can connect, load web page and login works, but if i click on anything its extremely slow if the page loads at all.

Log lines that concern me but i can not find anything on google that refers to the project.

2017-07-04 10:18:59: (mod_fastcgi.c.2702) FastCGI-stderr: PHP Warning:  fopen(/etc/raspap/raspap.auth): failed to open stream: No such file or directory in /etc/raspap/raspap.php on line 8
2017-07-04 10:18:59: (mod_fastcgi.c.2702) FastCGI-stderr: PHP Notice:  Undefined index: page in /var/www/html/index.php on line 51
2017-07-04 10:18:59: (mod_fastcgi.c.2702) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 33
2017-07-04 10:19:06: (mod_fastcgi.c.2702) FastCGI-stderr: PHP Warning:  fopen(/etc/raspap/raspap.auth): failed to open stream: No such file or directory in /etc/raspap/raspap.php on line 8

rovincent, 2017-07-04:
Actually i also get an undefined offset: 1 in dashboard when loading the dashboard ?
ending with undefined variable: wlan0up ?

wlan0 is the name of the interface.

rovincent, 2017-07-04:
When i do get the client config page to open i get the following error.

Undefined variable: networks in configure_client.php

The page seems to load and i have the list of available networks but when i configure the pass phase in any of the networks the {Add} button does not highlight and i can not click on it.

rovincent, 2017-07-04:
Interesting enough, other than very slow page loads and failures everything seems to work in the back end. Id sucessfully  write the wpa_supplicant file ?

still getting the raspap.auth file issue though and there is no file generated ?

One issue i found is that if the passphase has a - (dash) in the pass phase the add button does not load both the router and my phone were ok with the dash though ?

I removed it and the Add now works but page still rearly loads.

rovincent, 2017-07-04:
Ok, figured out that the FastCGI-stderr: PHP Warning: fopen(/etc/raspap/raspap.auth): failed to open stream: No such file or directory in /etc/raspap/raspap.php on line 8

disappears as soon as i entered a new password in the auth page of the ap.

SirLagz, 2017-10-03:
All of these warnings have now been resolved. They came up because sometimes the backend scripts could not find the data, but the front end still tried to access them. I've put some error checking in now to resolve that.

closed 2017-10-03 by SirLagz