RaspAP/raspap-webgui #90: Add support for Arch linux ARM on raspberry pi 3

hillz2, 2017-08-12:
I notice that raspap-webgui is for raspbian since there are apt-get and stuff. I'm running arch linux ARM on raspberry pi 3, can you please tell me which files I have to modify to get raspap-webgui work well with arch linux ? Thank you.

SirLagz, 2017-08-17:
apt was just to install packages.
You can install hostapd and git via pacman. Just need to install a DHCP server, web server, and PHP afterwards. Your choice of DHCP server and Web server.

You'll need to modify /etc/sudoers as per the manual install instructions,  but you'll need to ensure that the user specified is the same user as what the Web Server runs under.

/var/www/html should be the root of your web server, so if you're using a different web server/document root, update that as you need to.

Shouldn't be too much changing between Raspbian and Arch

billz, 2017-09-06:
@hillz2 @SirLagz closing this issue due to inactivity. Feel free to reopen if needed. Thanks.

closed 2017-09-06 by billz