RaspAP/raspap-webgui #91: Can RaspAP be used for headless bootstrapping of Wifi?

bluephlame, 2017-08-24:
Hi, I"m looking for a way to configure a Pi to connect to a wifi network in a headless way Can RaspAp support this? Or can it be easily adjusted to support this?

The process would be
- boot up the Pi, 
- it cannot connect to an existing wifi configuration (or doesn't have a configuration).
- Turn on the Acces point mode.
- Configure the Wifi Credentials for a local WiFi network.
- Reboot and connect to the wifi network (Acces point mode is switched off).

Thanks!

jspraul, 2017-09-13:
#77

SirLagz, 2017-10-03:
Yes, that is possible, but not yet implemented.
You could potentially implement it yourself with this script - http://sirlagz.net/2013/01/22/script-starting-hostapd-when-wifi-goes-down/

billz, 2018-10-14:
Added to the todo list. Closing this for now, thanks

closed 2018-10-14 by billz