RaspAP/raspap-webgui #93: Rtc does not work any longer after installing raspap

KeesvanUnen, 2017-08-30:
Yesterday I installed raspap. After that it seems that my RPI does not see the rtc any more. The /dev/rtc is missing. The file is back after reinstalling the rtc but is missing again after boot. Before installing raspap it worked fine. Any idea what causes this? Thanks.

billz, 2017-08-30:
The RaspAP installer doesn't touch the RTC or anything with the kernel config. Did RTC work after boot prior to installing RaspAP? If not, have you configured the service to start on boot, like so?

`update-rc.d hwclock.sh enable`

Try the above, reboot your pi and check the service with `hwclock -r --debug` and `service --status-all`

KeesvanUnen, 2017-08-30:
Yes, before that it worked fine after boot. I reinstalled the rtc again (/dev/rtc is present) and used the update-rc.d command. After reboot the /dev/rtc file is missing again. Hwclock -r--debug shows that /dev/rtc is missing. service --status-all shows: 
alsa-utils
avahi-daemon
bluetooth
console-setup.sh
cron

and seems to hang.

billz, 2017-08-31:
The RaspAP installer _does_ setup IP forwarding and NAT routing in `/etc/rc.local`. It's possible your rtc install also writes to this file, which could result in the device failing to start on boot. 

With rtc working, check the contents of this file with `sudo nano /etc/rc.local`. Depending on your hardware and Raspbian version, you may see something like:

```
echo ds1307 0x68 > /sys/class/i2c-adapter/i2c-0/new_device
```

If this is the case, try adding the contents of [this file](https://github.com/billz/raspap-webgui/blob/master/config/rc.local) to rc.local above the line `exit 0` and reboot your pi.

KeesvanUnen, 2017-09-01:
Hi Billz,

I have added:
dtoverlay=i2c-rtc,ds3231
in config.txt and this seems to have done the trick: no problems with the rtc any more. Thanks for your help. This thread can be closed.

closed 2017-09-01 by KeesvanUnen