RaspAP/raspap-webgui #95: RaspAP dashboard

KeesvanUnen, 2017-09-01:
Perhaps a silly question: how do you start the RaspAP gui dashboard?

Owiowi, 2017-09-01:
Not sure this is the answer you're looking for but if I understood your question correctly, you simply need to enter your raspberry pi IP address in your browser and then enter the default credentials (admin:secret)

KeesvanUnen, 2017-09-01:
I expected to find a RaspAP gui dashboard on my rasp-pi desktop somewhere. But I cannot find it. Can you tell me where I can enter the pi IP-address?

Owiowi, 2017-09-01:
You can access it from any device connected to your network by typing the IP address in your internet browser. E.g. On your computer, open your browser (firefox, chrome...) and type the raspberry pi address in the address bar.

KeesvanUnen, 2017-09-01:
Tried that, with other raspberry and tablet. There is no Wifi connection. Only thing I have done is downloading RaspAP with wget and reboot it. No configuration. Did I forget something?

billz, 2017-09-02:
@KeesvanUnen when you installed RaspAP with wget, the IP address or hostname you ssh'd into should now also host a web service. Without WiFi configured, and assuming your Pi is connected via ethernet, you can access RaspAP as @Owiowi suggested. 

If in doubt of your IP, use `sudo ifconfig`. The entry `inet addr:` under **eth0** is your Pi's IP address. You can also try `hostname -I`. Your router's firmware should also list the IP addresses of connected devices. Any of these methods will work.

closed 2017-09-02 by billz