RaspAP/raspap-webgui #96: Getting this to work as Wi-Fi repeater with an Edimax AC 600 (rtl 8812)

WStork, 2017-09-02:
Hi!
I know this is pretty much the "support multiple wireless interfaces" issue plus the "support static address on eth0" issue, but, well, the project I'm working with (we're ecologists, not IT people) is in a bit of a pickle.

Long story short, we needed a wireless access point like device that would be providing an access point with ssid "X" (where X is a unique number string) with DHCP and had a separate wireless interface (separate device) with a directional, powerful antenna for making outbound connections to other wifi devices (also DHCP, but this time working as client)

And a dedicated ethernet connection for managing these devices (with static IP)

Theoretically something like "raspberry pi + Edimax 600" fits this bill (and so does "generic access point with openwrt + Edimax 600, assuming you can get the driver work)

So, our sponsor just sent us a bunch (>50) rasberries 3 and edimax so we can start working (no, they did not check if drivers and/or wifi repeater configuration tools are available and work well, um... yeah)

So, we've spent a week (okay, I spent a week because I am the only person on the team who can at least try "compiling a driver on linux" and incidentally the only one who speaks English semi-decently) trying to get this whole thing off the ground using one of the rasberry pi 3 and one of edimaxes

It's not going along nicely (as in, there are numerous kinks and nothing works as expected)

We can't use a different dongle (I've barely compiled the driver for edimax on raspberry, and it is having issue when working with wpa_supplicant, as in, wpa-cli and thus raspap-webgui can't reconfigure it) because, well, they already sent us a box of these and there are only so many USB dongles with above-average, directional antennas that can do 5Ghz and have any linux support at all.

And moving from raspberry pi 3 is also problematic (because, well, we have box of those too and driver situation for rtl 8812au on OpenWRT seems even more dire)

So, some handholding in setting up the system would be greatly appreciated.

All we basically need is a wifi repeater that maintains a hotspot using raspberry pi 3's wifi, allows SSH (and only SSH) on eth0, connects as client to other wifi hotspots using an AC600 dongle, and allows forwarding traffic (including browsing, though browsing is not what it'll be used for) between raspberry's wifi and edimax AC600.

It seems like a prime raspap usecase.

We do have a bit of spare budget (we're not rich, so it's like, on the order of hundred bucks) to compensate for kind help.

And we can have our sponsor send the developers an Edimax AC600 for testing free of charge :)

SirLagz, 2017-10-01:
If this is still an issue, I can attempt to assist if I can get the dongle to test with.

SirLagz, 2017-11-02:
@WStork Is this still an issue? If not then I'll close this issue off.

closed 2017-11-16 by SirLagz