RaspAP/raspap-webgui #100: Add network function

sv6nob, 2017-09-10:
Hello,
Is there any way to add new networks on Configure client tab? I can see only saved networks with passwords, but i can't scan the available networks or add a new one manually.
Check the image https://ibb.co/kqjS1v
Thank you!

royord, 2017-09-15:
If you just refresh the page it seems to show the other networks that are available in your area, if you've connected to all of them I do believe that you would only see the ones that you have saved.

billz, 2017-09-21:
@sv6nob indeed @royord's comment is correct. Perhaps start a hotspot with your mobile device or try moving to an area with more wifi signals.

kfordaccela, 2017-11-10:
@billz @sv6nob this was fixed in commit "34c22e4"

billz, 2017-11-16:
@sv6nob closing this for now. Feel free to reopen if this isn't resolved for you, thanks.

closed 2017-11-16 by billz