RaspAP/raspap-webgui #102: WiFi config lost after installing raspap-webgui on a freshly installed Pi Zero W

T-vK, 2017-09-19:
I have Pi Zero W. I just flashed the latest Raspbian Lite on an a Micro SD card. All I changed was: 

 - I enabled ssh
 - I changed the hostname
 - I added a WiFi network to auto connect to

If you want to know what exactly I did, I called these functions:

``` bash
enableSsh () {
    sudo touch "${bootPath}/ssh"
}

changeHostname () { # changeHostname(string hostname)
    newHostname=$1
    oldHostname=`sudo cat ${rootPath}/etc/hostname`
    sudo echo "${newHostname}" | sudo tee "${rootPath}/etc/hostname" > /dev/null
    sudo sed -i -e "s/${oldHostname}/${newHostname}/g" "${rootPath}/etc/hosts"
}

setupWifiClient () { # setupWifiClient(string ssid, string password)
    ssid=$1
    password=$2
    echo "
network={
    ssid=\"$1\"
    psk=\"$2\"
}" | sudo tee -a "${rootPath}/etc/wpa_supplicant/wpa_supplicant.conf" > /dev/null
}
```

Then I inserted the card into the Pi Zero W for the first time. And as expected it automatically connected to my WiFi and I was able to ssh into it using the new hostname.  
Then I installed `raspap-webgui` using `wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap`. I said yes to everything and after the reboot it did not connect to my WiFi again.  

So I can't access the web interface. I can't even ssh into the Pi again.  
I checked the card on my computer. The config for my network is still in `sdcard/etc/wpa_supplicant/wpa_supplicant.conf`.  But why doesn't it connect anymore?  

I can't just use HDMI because I don't have an HDMI monitor or adapter. 

So how do I install raspap-webgui over ssh in a way that I can access it after the installation?  

I have tried to get USB Ethernet working, but after trying 3 different tutorials I just gave up. There must be another way!

SirLagz, 2017-10-03:
RaspAP uses it's own IP configuration by default. This will be changed in later versions to potentially use DHCP initially

billz, 2019-04-21:
@T-vK the Pi Zero W was only partly supported when this issue was created. The project has had many updates since then specifically targeted at this device. Suggest you try the Quick Installer again. 

You might also try USB on-the-go (OTG) with the Pi Zero as this greatly simplifies ssh access, particularly when network configs are changing.

billz, 2019-08-17:
Closing this as the project now fully supports managed mode APs.

closed 2019-08-17 by billz