RaspAP/raspap-webgui #106: SaveTORProxySettings back to /etc/tor/torrc file

brunotvaz, 2017-09-24:
Hi Billz,

First of all let me thank you for this SUPER amazing tool. Awesome and easy to use. Thank you !
I need help with saving the settings on the TOR PAGE back to the TORRC file. I've enabled TOR on the settings as mentioned in the README file:
**1st - Enable TOR PROXY:** (/var/www/html/includes/index.php)
changed: define('RASPI_TORPROXY_ENABLED', false );
to: define('RASPI_TORPROXY_ENABLED', true );
**2nd - SHOW settings from torrc file:**
changed the /var/www/html//includes/functions.php file ([functions.php.txt](https://github.com/billz/raspap-webgui/files/1327199/functions.php.txt)) to show settings of torrc file:
added: ExitNodes, StrictNodes and TransPort settings
also added a button to list iptables. This prints  the output of: iptables -t nat -S for viewing purposes only.
After that I had to change /etc/sudoers file to allow "my custom commands" to be executed with the 
www-data user:
```
www-data ALL=(ALL) NOPASSWD:/etc/init.d/tor start
www-data ALL=(ALL) NOPASSWD:/etc/init.d/tor stop
www-data ALL=(ALL) NOPASSWD:/sbin/iptables -t nat -S
www-data ALL=(ALL) NOPASSWD:./home/pi/toron.sh
www-data ALL=(ALL) NOPASSWD:./home/pi/toroff.sh

```Notice the scripts /home/pi/toron.sh and /home/pi/toroff.sh. I've changed the default commands (**StartTOR** , **StopTOR**) because I needed to restart TOR and change IPTABLES rules at the same time hence the need to create custom scripts.

```
#!/bin/bash
# /home/pi/toroff.sh
/etc/init.d/tor stop
iptables -t nat -D PREROUTING -i wlan1 -p tcp --dport 8080 -j REDIRECT --to-ports 8080
iptables -t nat -D PREROUTING -i wlan1 -p tcp --dport 22 -j REDIRECT --to-ports 22
iptables -t nat -D PREROUTING -i wlan1 -p udp --dport 53 -j REDIRECT --to-ports 53
iptables -t nat -D PREROUTING -i wlan1 -p tcp --syn -j REDIRECT --to-ports 9040
```

```
#!/bin/bash
# /home/pi/toron.sh
/etc/init.d/tor start
iptables -t nat -A PREROUTING -i wlan1 -p tcp --dport 8080 -j REDIRECT --to-ports 8080
iptables -t nat -A PREROUTING -i wlan1 -p tcp --dport 22 -j REDIRECT --to-ports 22
iptables -t nat -A PREROUTING -i wlan1 -p udp --dport 53 -j REDIRECT --to-ports 53
iptables -t nat -A PREROUTING -i wlan1 -p tcp --syn -j REDIRECT --to-ports 9040
```

**_NOTE:_** Due to port forwarding issues when enabling TOR I was forced to change the lighttpd port to 8080, otherwise all visited webpages would be redirected to the RaspAP webpage.
Inside /etc/lighttpd/lighttpd.conf
changed: **server.port                 = 80**
to: **server.port                 = 8080**
Then restart it by running: **sudo service lighttpd restart**
What I need help with is a way to save the settings on the TOR PAGE back to the TORRC file.
Can someone please help me accomplish that? Many thanks in advance.

wkpn, 2019-05-18:
@brunotvaz can you share your torrc config please?

billz, 2025-03-22:
This would be a _great_ candidate for a [custom user plugin](https://docs.raspap.com/custom-plugins/).