RaspAP/raspap-webgui #107: Uninstall script

art049, 2017-09-28:
Hello, I want to uninstall raspap.
Do you have a script to do it cleanly ?

SirLagz, 2017-10-03:
Not yet. It's in progress now.

SirLagz, 2017-10-03:
This has been created now. Just needs some willing testers!

billz, 2017-10-03:
Tested on fresh install of Jessie Lite.

Restoring backups for interfaces, hostapd, dnsmasq, dhcpcd.conf all successful.

Install directory `/etc/raspap` removed. Lighttpd directory `/var/www/html` unchanged. Should be able to fix like so:

```
sudo rm -rf "$webroot_dir"/*
```

SirLagz, 2017-10-03:
Thanks Bill. I've updated the uninstall script.

closed 2017-10-21 by SirLagz

GunnarKammel, 2019-05-01:
how can I uninstall raspap?

billz, 2019-05-02:
@GunnarKammel this is covered in the [FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#how-do-i-remove-raspap).

Festerstoybox, 2021-01-24:
i have tried to uninstall using the FAQs option but i get the error saying "no such file or directory" and when i go to that folder in gui there are no files to use (installer/uninstaller).
not sure why this is happening as i installed using the guide.