RaspAP/raspap-webgui #108: Backup configuration file before installing

art049, 2017-09-28:
I've lost my whole pi configuration installing this.
Performing a backup of configuration files is necessary during the installation.

SirLagz, 2017-10-01:
Which configuration file?

art049, 2017-10-01:
/etc/network/interfaces
Hostapd configuration also.

Youre code simply overwrite without even asking.

SirLagz, 2017-10-01:
Thanks for that. We'll look into this issue.

SirLagz, 2017-10-01:
I've added a section into the installer to back up /etc/network/interfaces and /etc/hostapd/hostapd.conf files if found. They are moved to /etc/raspap/backups.

closed 2017-10-01 by SirLagz

art049, 2017-10-01:
Thanks for fixing that.
I think it's also needed for  /etc/dnsmasq.conf and /etc/dhcpcd.conf

SirLagz, 2017-10-01:
Yep, newest version already has logic for those 2 files too, will be up soon