RaspAP/raspap-webgui #109: Broken with the latest stretch image (tested on a fresh install)

T-vK, 2017-09-30:
every time:    
 - I flashed `2017-08-16-raspbian-stretch` onto the card and confirmed that the data matches and is not corrupted.  
 - I put the card into the Raspberry Pi Zero W
 - I plugged in my keyboard, mouse and HDMI
 - I plugged in the power
 - I waited for it to boot
 - When it was done I landed on the desktop automatically, no login was required.
 - I clicked the WiFi icon at the top right corner and connected to my WPA2 home WiFi.
 - I opened the browser and went to the Github page to copy the install command
 - I opened a new terminal using the icon at the top of the screen.
 - In that terminal I ran the install command: `wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap`
 - I said yes to everything, even the reboot.
 
After the reboot, I am no longer able to resolve dns queries. I am connected to my WiFi, though. 
I can access localhost from the browser and log in. I checked the Access Point config and it said `hostapd is not running`. I changed the channel to 11 (just like my home WiFi) and tried to start it again, but still `hostapd is not running`. 

I checked my sudoers file and it does contain these lines, so everything should be fine there:

```
www-data ALL=(ALL) NOPASSWD:/sbin/ifdown wlan0
www-data ALL=(ALL) NOPASSWD:/sbin/ifup wlan0
www-data ALL=(ALL) NOPASSWD:/bin/cat /etc/wpa_supplicant/wpa_supplicant.conf
www-data ALL=(ALL) NOPASSWD:/bin/cp /tmp/wifidata /etc/wpa_supplicant/wpa_supplicant.conf
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli scan_results
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli scan
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli reconfigure
www-data ALL=(ALL) NOPASSWD:/bin/cp /tmp/hostapddata /etc/hostapd/hostapd.conf
www-data ALL=(ALL) NOPASSWD:/etc/init.d/hostapd start
www-data ALL=(ALL) NOPASSWD:/etc/init.d/hostapd stop
www-data ALL=(ALL) NOPASSWD:/etc/init.d/dnsmasq start
www-data ALL=(ALL) NOPASSWD:/etc/init.d/dnsmasq stop
www-data ALL=(ALL) NOPASSWD:/bin/cp /tmp/dhcpddata /etc/dnsmasq.conf
www-data ALL=(ALL) NOPASSWD:/sbin/shutdown -h now
www-data ALL=(ALL) NOPASSWD:/sbin/reboot
```
The I checked this:

```
$ sudo -u web-data sudo /etc/init.d/hostapd start
sudo: unknown user web-data
sudo: unable to initialize policy plugin
```

And this: 

```
$ sudo cat /var/log/lighttpd/error.log
2017-09-30 14:41:35: (log.c.217) server started
2017-09-30 14:43:38: (server.c.1828) server stopped by UID = 0 PID = 1
2017-09-30 14:43:40: (log.c.217) server started
2017-09-30 14:49:08: (server.c.1828) server stopped by UID = 0 PID = 1
2017-09-30 14:50:30: (log.c.217) server started
2017-09-30 14:51:30: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Warning:  fopen(/etc/raspap/raspap.auth): failed to open stream: No such file or directory in /etc/raspap/raspap.php on line 8
2017-09-30 14:51:30: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined index: PHP_AUTH_USER in /var/www/html/includes/authenticate.php on line 2
2017-09-30 14:51:46: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined index: PHP_AUTH_PW in /var/www/html/includes/authenticate.php on line 3
2017-09-30 14:51:46: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Warning:  fopen(/etc/raspap/raspap.auth): failed to open stream: No such file or directoryin /etc/raspap/raspap.php on line 8
2017-09-30 14:51:46: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined index: page in /var/www/html/index.php on line 51
2017-09-30 14:51:46: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 19
2017-09-30 14:51:46: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 21
2017-09-30 14:51:46: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 23
2017-09-30 14:51:46: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 25
2017-09-30 14:51:46: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 27
2017-09-30 14:51:46: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 29
2017-09-30 14:51:46: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/dashboard.php on line 31
2017-09-30 14:51:19: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Warning:  fopen(/etc/raspap/raspap.auth): failed to open stream: No such file or directory in /etc/raspap/raspap.php on line 8
2017-09-30 14:51:41: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Warning:  fopen(/etc/raspap/raspap.auth): failed to open stream: No such file or directory in /etc/raspap/raspap.php on line 8
2017-09-30 14:53:10: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Warning:  fopen(/etc/raspap/raspap.auth): failed to open stream: No such file or directory in /etc/raspap/raspap.php on line 8
2017-09-30 14:53:10: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined offset: 1 in /var/www/html/includes/hostapd.php on line 60
2017-09-30 14:53:30: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Warning:  fopen(/etc/raspap/raspap.auth): failed to open stream: No such file or directory in /etc/raspap/raspap.php on line 8
```

SirLagz, 2017-10-01:
`$ sudo -u web-data sudo /etc/init.d/hostapd start
sudo: unknown user web-data
sudo: unable to initialize policy plugin`

You should be using `www-data` and not `web-data` there. Try that and see what comes up.

T-vK, 2017-10-01:
Okay here is what I did:

```
$ sudo -u www-data sudo /etc/init.d/hostapd start
[ ok ] Starting hostapd (via systemctl): hostapd.service.
```

```
$ sudo -u www-data sudo /sbin/ifup wlan0
ifup: Unknown interface wlan0
```

```
$ ifconfig
lo: flags=73<UP,LOOPBACK,RUNNING> mtu 65536
        inet 127.0.0.1 netmask 255.0.0.0
        inet6 ::1 prefixlen 128 scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 1440  bytes 96480 (94.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1440  bytes 96480 (94.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.3.141.1  netmask 255.255.255.0  broadcast 10.3.141.255
        inet6 fe80::1cad:b0ef:5cd3:e19b  preflen 64  scopeid 0x20<link>
        ether b8:27:eb:09:af:5a
        RX packets 33  bytes 5786 (5.6 KiB)
        RX errors 0  dropped 0 overruns 0  frame 0
        TX packets 45  bytes 6506 (6.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions
```

Then I opened up the browser again and checked if I tried to open github.com, without success. Then I opened localhost and logged in again. I went to the `configure hotspot` site again and clicked on `start hotspot` again, but it still reports `hostapd is not running`.  

I should mention that it's not only dns queries for internet addresses that fail, I can't even ping 8.8.8.8.  
I also scanned for available WiFi networks from another computer and sure enough the Raspi was not hosting a hotspot.

SirLagz, 2017-10-01:
What's the output of `wpa_cli status` ?

T-vK, 2017-10-01:
```
$ wpa_cli status
Selected interface 'p2p-dev-wlan0'
wpa_state=DISCONNECTED
p2p_device_address=b8:27:eb:09:af:5a
address=b8:27:eb:09:af:5a
uuid=9e76d37-3118-52a2-80f2-61f6262d1ba3
```

SirLagz, 2017-10-01:
Ok, so the WiFI is disconnected. Probably because you tried to start hostapd.

Hostapd should not be started unless you *want* it to broadcast.
If you're just trying to set it to connect to your wifi, then you should be able to configure the client through the web interface and tell it to connect.

T-vK, 2017-10-01:
Well, I want it to connect to my home WiFI and at the same time host a hotspot. 
If that's not possible (but it should be), it should just host a hotspot.  

The desktop as well as the webgui dashboard claim that I am connected to my home WiFi.

SirLagz, 2017-10-01:
Which WiFi NIC are you using?

T-vK, 2017-10-01:
The internal one. (It's a Raspberry Pi Zero W)

When I check with iwconfig btw, it does say that I'm connected to my home WiFi. (I removed the SSID and MAC for privacy reasons.)

```
$ iwconfig
wlan0      IEEE 802.11  ESSID:"xxxxxxxx"
           Mode:Managed  Frequency:2.462 GHz  Access Point: XX:XX:XX:XX:XX:XX
           Bit Rate=24 Mb/s   Tx-Power=31 dBm
           Retry short limit:7   RTS thr:0ff   Fragment thr:off
           Power management:on
           Link Quality=70/70  Signal level=-40dBm
           Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
           Tx excessive retries:0  Invalid misc:0   Missed beacon:0

lo         no wireless extensions.
```

SirLagz, 2017-10-01:
Ah. RaspAP has not been coded to allow for both client and access point mode on a Pi Zero.
In this case, what happens when you try to set it up as a hotspot?

T-vK, 2017-10-01:
A few more things:

 - The `disable wlan0` in the dashboard doesn't work.
 - When I delete my home WiFi from the client site, I'm still connected to my home WiFi.
 - When I restart WiFi using the WiFi icon in the top right corner of the screen, I'm still connected to my home WiFi.
 - When I try to use the icon to disconnect from my home WiFi, nothing happens.

The entry for my home WiFi has been removed successfully from `/etc/wpa_supplicant/wpa_supplicant.conf`. I assume the webgui just failed to reload a service or something like that.  

`raspap-webgui` is clearly broken if you ask me. I mean look at the PHP warnings and notices from the first post. Something is not right there.

SirLagz, 2017-10-01:
I've just noticed that 'Stop wlan0' doesn't work as well.
It's been a while since I've worked on, or even used RaspAP myself, but I'm getting back into it so hopefully these issues will be gone soon.

Those PHP warnings are the first thing that I've fixed, so the next update of RaspAP should be a bit better :)

T-vK, 2017-10-01:
You say you have already fixed the php warnings. Is that version available somewhere?

SirLagz, 2017-10-01:
About to push it up now.

T-vK, 2017-10-01:
Awesome! 
There doesn't seem to be an uninstaller. Can I just install it over the old version?

SirLagz, 2017-10-01:
Yep, just run the install script again and it'll move the old installation out of the way.

T-vK, 2017-10-01:
Oh, I just realize because I don't have internet access on the Raspi anymore, I can't run the wget command...

T-vK, 2017-10-01:
It fails with an installation error saying:  `RaspAP install error: Cannot enable fastcgi-php for lighttp`

T-vK, 2017-10-01:
It fails with an installation error saying: 

```
RaspAP Install: Enabling PHP for lighttpd
already enabled
Run "service lighttpd force-reload" to enable changes
RaspAP Install Error: Cannot enable fastcgi-php for lighttpd
```

SirLagz, 2017-10-01:
I've fixed up the install file to allow in-place reinstalls now.

T-vK, 2017-10-01:
Great, I'll try a reinstall now.

SirLagz, 2017-10-01:
Right. A few more issues still. I'll keep plugging away at it.

SirLagz, 2017-10-21:
@T-vK I think I've fixed up some more issues. When you have a chance, can you please give it a shot and report back on the errors that remain? Thanks.

T-vK, 2017-10-22:
@SirLagz 
I set up a fresh Raspbian installation and tried to install RaspAP, but I get an error:

`E: Package 'php5-cgi' has no installation candidate`

SirLagz, 2017-10-23:
Sounds like Raspbian has updated package availabilities...I installed it on the same image earlier. We'll need to look into this further.

billz, 2017-11-16:
@T-vK the installation issue should be resolved now with #124. Please update and try again. Thanks.

billz, 2019-04-13:
Closing due to inactivity. See also #297 which addresses this specific issue with the Pi Zero W.

closed 2019-04-13 by billz