RaspAP/raspap-webgui #112: update script

nunziofiore, 2017-10-03:
Hi All, sorry if my question could be newbie or you have an ansower yet in some thread. 
I read that you're working on a UNINSTALL SCRIPT and this is great.
I'm using this project as a base for my custom project and it's very useful. I 'm thinking to release some open source feature if you'd ike, about custom theme and others stuff I'm working.
My question is about an UPDATE SCRIPT.
FOr example, you now release the UNINSTALL SCRIPT, it could be useful to update our installations to have it. Also could be interesting to have an update everytime there's some coool stuff in a new release. In my custom project we have a webservice based on a version json that it's checked from the UI, when there's something to update  we have a signal on the UI and clicking on the link the user can update it's custom version of software, db and scripts.

SirLagz, 2017-10-03:
At the moment, if you re-run the install script/command, it will grab the newest version and move all the old files out of the way, so in a way, reinstalling and updating will do the same thing.

nunziofiore, 2017-10-03:
Ok nice. The install script has some error. You have to disable lighty-enable-mod fastcgi-php
before launch the script, otherwise the installation stop.
Thank you
Nunzio

SirLagz, 2017-10-03:
The newest version fo the install script should've fixed that...I'll double check it.

nunziofiore, 2017-10-03:
great

SirLagz, 2017-10-03:
I tried installing with the command `wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap` and it worked fine. What command are you using to reinstall/update?

nunziofiore, 2017-10-03:
I tried on half of September and that command broken because fastcgi was yet enabled. I'll have to retry in next days, thank you

SirLagz, 2017-10-03:
Ah, I only fixed it a couple of days ago.

SirLagz, 2017-10-14:
@nunziofiore Have you tried the new installer? Any Issues? If none then I'll close this issue.

closed 2017-10-18 by SirLagz

nunziofiore, 2017-11-03:
I confirm that now it's ok!