RaspAP/raspap-webgui #116: Cannot Change Admin Password

jrjohnson1231, 2017-10-09:
System: Pi 3 Model B

I have been able to change all of the default configurations except the admin password. I get the error "Failed to update admin password"

I checked the lighttpd logs and see:

`PHP Warning: fopen(): Filename cannot be empty in /var/www/html/includes/admin.php on line 16`

BenoitDLX, 2017-10-12:
Hello, i've found the problem.

It looks like when you try to edit the password, the script is looking for the following file: `/etc/raspap/raspap.auth` but it doesn't exist.

When you create it, the website doesn't return an error when you edit your password but in fact, you can't connect anymore and the lighthttpd.log returns:

`FastCGI-stderr: PHP Notice:  Undefined index: PHP_AUTH_PW in /var/www/html/includes/authenticate.php on line 3`

SirLagz, 2017-10-13:
That `undefined index` error just means that PHP can't find the `PHP_AUTH_PW` variable in the file.
You need to make sure the contents of `/etc/raspap/raspap.auth` are valid, but there should also be some error checking to make sure that the contents of it are valid, or otherwise ignore the file. Something to be worked on.

jrjohnson1231, 2017-10-13:
Currently the file does not exist on my system, is there a default one that was supposed to be made on installation/setup?

SirLagz, 2017-10-13:
If it doesn't exist then the default username / password is used.

jrjohnson1231, 2017-10-13:
I understand that, but I want to change from the default and currently cannot, which seems to be caused by that file not existing.

SirLagz, 2017-10-13:
Ah, I just realised why.
`define('RASPI_CONFIG', '/etc/raspap');
if(file_exists(RASPI_CONFIG.'/raspap.auth')) {
    define('RASPI_ADMIN_DETAILS', RASPI_CONFIG.'/raspap.auth');
} else {
    define('RASPI_ADMIN_DETAILS','');
}`
The code unsets `RASPI_ADMIN_DETAILS` if the file doesn't exist. This needs to be fixed, whether it's creating it by default or something else, I'm not sure yet.

commit a24d3d8f (2017-10-14): Moved RASPI_ADMIN_DETAILS file check to the admin.php page which will now create the file if it doesn't exist - to resolve #116. Commented out check on index.php
closed 2017-10-14 by SirLagz