RaspAP/raspap-webgui #118: Support OrangePI

heX16, 2017-10-11:
I try install to OrangePi Zero.

```ShellSession
RaspAP Install: Installing required packages
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package php5-cgi is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'php5-cgi' has no installation candidate
RaspAP Install Error: Unable to install dependencies

```

SirLagz, 2017-10-11:
What distribution are you using?

lkoepsel, 2017-10-22:
Hi, I had the same issue using the latest version of raspbian...

lkoepsel, 2017-10-22:
sorry, hit the wrong key: 2017-09-07-raspbian-stretch-lite. I tried to manually install, using just php-cgi. That worked, everything to install, however, I wasn't succesfull with the rest of the manual install.
Does raspap-webgui require PHP5 instead of PHP7?
Thanks,
Lief

SirLagz, 2017-10-23:
Raspbian seems to have changed the availability of the PHP packages, we need to have a further look into this.

commit 6e2c6de8 (2017-11-16): Merge pull request #124 from billz/new-php-pa

Installs php7.0-cgi for Stretch. Maintains backward compatibility with Jessie. Resolves #118 #109
closed 2017-11-16 by billz