RaspAP/raspap-webgui #120: Installation overwrites rc.local

jadero, 2017-10-15:
I did an install today and realized later that my TVheadend Kodi client couldn't connect anymore to the server. A quick check showed that my custom made entries to rc.local were gone and an iptables entry was there instead which hasn't been there before I installed raspap. I couldn't find any backup file so this took me a while to get everything running again. :^(

jadero, 2017-10-15:
Btw. I followed this tutorial for installation: [https://howtoraspberrypi.com/create-a-wi-fi-hotspot-in-less-than-10-minutes-with-pi-raspberry/](url).

SirLagz, 2017-10-18:
There should be backups stored in /etc/raspap/backups. Did you check there?

jadero, 2017-10-18:
After some testing I decided to remove raspap and all dependencies incl. folders in /etc. When I discovered that there was something wrong with my rc.local the raspap folder in /etc was already gone so I cannot answer your question. So I suggest that you put the backup file for rc.local right into /etc (append a suffix to it). But the question remains why there where lines removed in rc.local in the first place?

SirLagz, 2017-10-18:
The rc.local file is updated to add in the rules for the RaspAP web GUI to act as a router.
Uninstalling RaspAP with the uninstall script would have restored the rc.local file that you had before.

jadero, 2017-10-18:
Ok thats fine concerning the uninstallation. I didn't know that there is a script available though. But I would still consider it as a bug that custom added lines to the rc.local script are removed during installation (or that this file is replaced, whatever).

SirLagz, 2017-10-18:
Yep, I agree. It is an issue we need to address.

commit 6d902934 (2017-10-22): Merge pull request #123 from SirLagz/master

Updated Install/Uninstall scripts to resolve #120
closed 2017-10-22 by SirLagz